RaspAP/raspap-webgui #377: Invalid CSRF TOKEN

JustACodingFox, 2019-08-14:
### Subject of the issue
If I press the "stop wlan0" button or if I try to connect to a wifi, I get the error "Invalid CSRF TOKEN". 

### Your environment
* Raspberry Pi hardware:  Pi Zero W
* Raspbian version: Raspbian GNU/Linux 10 (buster)
* Followed the project prerequisites? Yes
* Checked the project FAQ? Yes
* RaspAP Quick Install or Manual setup? Tried both. Resetted the system in between.
* Using default configuration? Yes
* Simultaneous AP and managed mode? I want to but i can´t.
* Other software or services running with RaspAP? Not that I know of

### Steps to reproduce
Follow installation instructions and try to stop wlan or to connect to a wifi.

### Expected behavior
It should connect to a wifi or stop wlan.

### Actual behavior
I get forwarded to a white website with the sentence "invalid csrf token"

JustACodingFox, 2019-08-15:
I installed rasbian buster lite fresh and updated + upgraded the system. After that I selected my locale and installed per quick install but the error still persist.

billz, 2019-08-16:
Merged PR #376. Could be related to what you reported, although I was unable to repro this on a Pi Zero W. Please update your install and try again, thanks

Slluxx, 2019-08-18:
still an issue. Rasp 3b+ when starting the client interface

PR #385 (2019-08-19): fix csrf token xhr race condition
fixes #377. the initial ajax-loading of wifi stations regenerated the token in the session so any future form submission (e.g., connecting to a network) would fail guaranteed. bummer. will look into a more solid implementation.
closed 2019-08-19 by billz