RaspAP/raspap-webgui #379: Allow eth0 and wlan0 bridge during install

sorinello, 2019-08-16:
This is just a feature request.

My use case is as follows:

1) I want my main DHCP server to issue IP address leases so I can better monitor the allocation from a central point. Alsi I want only one place to define static DHCP.
2) If eth0 and wlan0 will be bridged, there will be no NAT-ting, thus I expect a performance increase

I realize this might not be easy to support since the actual setup and my desired setup are quite different, but maybe during the installation an option can be requested from the user which mode he wants. The default can remain the current one.

billz, 2019-08-16:
Related to #378

sorinello, 2019-08-17:
As a weekend project, I started working on this, and managed to do a **functional** early alpha implementation.

Please note that this is **only headless**, CLI does **not work** !

Only testes on the latest Raspbian 10 - Buster. I uesd the Lite image.

If someone is willing to test me towards the implementation, please feel free, steps below:

1) Clean Raspbian 10 install
2) git clone https://github.com/sorinello/raspap-webgui
3) cd raspap-webgui/installers
4) ./common_bridged.sh

This will create a br0 bridge between eth0 and wlan0. The bridge will need a DHCP server in the LAN in order to obtain an IP address.

The SSID and password are the same as for the default instalation.

Since maybe one day we'll want to integrate/merge back in the origin repo, I have not touched any of the original files, allowing the official maintainers to see more easily the differences and integrate them how they see fit.

Depending if this current state will generate interest, I am willing to continue working on it and help to port it in the official repository. 
For the moment, I will let the maintainers decide if they want this merged, and if it sparks enough interest from the community.

4suino, 2019-08-21:
> As a weekend project, I started working on this, and managed to do a **functional** early alpha implementation.
> 
> Please note that this is **only headless**, GUI does **not work** !
> 
> Only testes on the latest Raspbian 10 - Buster. I used the Lite image.
> 
> If someone is willing to test me towards the implementation, please feel free, steps below:
> 
> 1. Clean Raspbian 10 install
> 2. git clone https://github.com/sorinello/raspap-webgui
> 3. cd raspap-webgui/installers
> 4. ./common_bridged.sh
> 
> This will create a br0 bridge between eth0 and wlan0. The bridge will **need a DHCP server in the LAN** in order to obtain an IP address. If you have a static DHCP, the DHCP will see the MAC of eth0 interface as the MAC of the br0.
> 
> The SSID and password are the same as for the default installation.
> 
> Since maybe one day we'll want to integrate/merge back in the origin repository, I have not touched any of the original files, allowing the official maintainers to see more easily the differences and integrate them how they see fit.
> 
> Depending if this current state will generate interest, I am willing to continue working on it and help to port it in the official repository.
> For the moment, I will let the maintainers decide if they want this implementation continued, and if it sparks enough interest from the community.

I really appreciate this feature, I've the same issue. Thank you.

glaszig, 2019-08-21:
that‘s a nice feature. i think anything network-related fits the scope of this platform. but all of this needs ui and should be optional because not everybody wants a bridge. maybe existing bridge configurations need to be detected as well.

my 2 cents:
- please develop in a seperate branch as it will be infeasable to merge your master if you go ahead and change anything you like on that branch
- rebase your branch onto the upstream master; helps keep the history readable
- please don‘t commit experiments like commenting things out or at least squash them at some later point

example: if we now merge your master, we get your repo as the installer source (btw, it‘d be a nice feature for developers to be able to override the git source by way of an env var)

> have not touched any of the original files, allowing the official maintainers to see more easily the differences

that‘s actually bad because that‘s what ˋgit diffˋ is for and now we have seperate files to digest.

anyway, if you need help let me know.

4suino, 2019-08-21:
> that‘s a nice feature. i think anything network-related fits the scope of this platform. but all of this needs ui and should be optional because not everybody wants a bridge. maybe existing bridge configurations need to be detected as well.

Yes, you have true. The feature can be implemented like some hardware router that let you choose the type of configuration/mode you want.

sorinello, 2019-08-27:
@glaszig  Thanks for your feedback. Can you elaboratre more on the following ?

1) 

> please develop in a seperate branch as it will be infeasable to merge your master if you go ahead and change anything you like on that branch

So on my own fork, should I create another branch ? I don't quite understand what difference it makes if I am using my master or another branch from my fork.

2) 

> please don‘t commit experiments like commenting things out or at least squash them at some later point

I think will be a bit harder not to do, since sometimes I have to use a trial and error strategy in order to progress.

Please give me a bit more info on 1) and 2). I could erase my current fork and start from scratch on another branch.

Currently, I am in a bit of a dilemma. What should be my strategy: 

1) Should I continue on the path I started and start to make new UI pages for this current implementation and work on merging the 2 solutions later
OR
2) Start integrating my script in the existing installer so that users can have an option to make a bridge during install, even if the UI is not yet working.

Thanks in advance !

4suino, 2019-11-07:
Hi all, is there any update on this request?

billz, 2019-11-08:
@sorinello nice feature. re: contributing to the project, [these steps](https://github.com/billz/raspap-webgui#how-to-contribute) will get you on the right track. 

I've recently merged a PR that extends the quick installer, so this would facilitate adding an optional bridge setup.

4suino, 2019-11-08:
Thank you, and sorry for the question. How can I update the currently setup and add this feature?

911pcdoc, 2019-12-22:
how does one change the ssid and password for the hotspot in this mode?

scoobaspeaz, 2020-02-13:
@sorinello so what did you end up doing? Your method is exactly what I am wanting to use. Bridge mode with eth0 and wlan0 seems to be the logical choice since this emulates how a store bought router behaves. Did you end up forking this (yours is an older v 1)  or are we waiting on @billz to include it in the latest branch?

Taikuh, 2020-03-05:
https://github.com/raspberrypi/documentation/blob/aaf97e8c5c85e1aeb33bdd5e4119e6330a2e2f51/configuration/wireless/access-point-bridged.md

Use the instructions above with a fresh install of Raspbian to be safe. The instructions above didn't work for me. I had to add an extra file:

```bash
# /etc/systemd/network/bridge-br0.network
[Match]
Name=br0

[Network]
DHCP=ipv4
```

I don't think anybody is working on this feature. I'm still trying to parse through `common.sh` and brushing up on my php to better decide how to move forward and add a Bridge vs Routed toggle.

billz, 2020-03-13:
Closing this in favor of #522. Thanks everyone for your contributions

closed 2020-03-13 by billz