RaspAP/raspap-webgui #380: Description of what things do in the webinterface (or somehwere else (wiki?))

Slluxx, 2019-08-17:
**Is your feature request related to a problem? Please describe.**
Might be. For exmample: what does choosing an interface mean in each area of the gui? somehow no setting i try is working. Literally nothing i do seems to have an effect.

**Describe the solution you'd like**
Clear descriptions. What does what.

**Describe alternatives you've considered**
-

**Additional context**
Also wlan0 and wlan1 switching are not working. wlan1 as client, wlan0 as hotspot.
ive read every single issue regarding this and i spend hours of try and error with no luck. Probably cause i have to use buster desktop to connect to my wifi which needs authentication (like in a hotel). doing stuff on the desktop probably messes with raspap. 

Sad, cause its the only hope i have to get better internet in my room. (dedicated ap's and extenders are not working in this wifi)

billz, 2019-08-18:
@Slluxx two of your best starting points are:

https://wireless.wiki.kernel.org/en/users/Documentation/hostapd 
https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf

Perhaps you noticed the helpful "Information provided by..." labels in the UI? Learning what these services are and how they work should demystify things.

> Also wlan0 and wlan1 switching are not working

The project does not claim to support this. At this stage it's very much a DIY affair.

billz, 2019-08-19:
Thanks for raising this @Slluxx. I'll update the wiki accordingly. Contextual help in the UI (eg, tooltips) would be nice, although this would require translating strings into 13 languages.

billz, 2019-08-27:
Added a new [FAQ item](https://github.com/billz/raspap-webgui/wiki/FAQs#settings) to address this, thanks

closed 2019-08-27 by billz