RaspAP/raspap-webgui #382: Switch wlan0 AP to external adapter

pulkin, 2019-08-18:
I was previously managing my pi3 networking through configuration files and it was fine most of the time. I thought that this project would ease up the process. Not the case, unfortunately.

**Describe the bug**
I set up `raspap` with the README one-liner. I do some basic setup until the AP works.

I then want to perform a simple thing: delegate all the AP stack to my external USB adapter because the pi3 built-in wifi [has serious issues](https://github.com/raspberrypi/linux/issues/2453). I go to "Configure hotspot" -> "Stop", choose the desired interface, then "Save" and "Start". I can no longer connect to the network.

The actual issue is, of course, due to some additional configuring needed.
log:
```
raspi dnsmasq-dhcp[2715]: no address range available for DHCP request via wlx0123456789AB
```
which is likely due to static IP being not set
```
wlx0123456789AB: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.239.40  netmask 255.255.0.0  broadcast 169.254.255.255
```
In addition, the "Dashboard" still shows `wlan0` as the interface of interest. Thus,

1. raspap does not manage essential interface settings automatically and consistently;
2. raspap does not provide a clear distinction between what I can manage in the web GUI and what **has** to be done manually;
3. searching for "wlan0" and "10.3.141.1" in the project folder gives an immediate idea that a whole bunch of settings are hard-coded. Unless you know how php projects are organized, it is absolutely not clear whether going to /etc/network/interfaces and setting up the interface there is the solution that will work in long term.

**Your environment**
* Raspberry Pi hardware Pi 3 Model B
* Raspbian version Buster Lite
* RaspAP Quick Install
* Simultaneous AP and managed mode? N
* Onboard wireless chipset or external adapter? both
* Other software or services running with RaspAP? almost clean install

pulkin, 2019-08-18:
I am investigating a wild idea that the variable `RASPI_WIFI_CLIENT_INTERFACE` is not **only** about the wifi client interface as it states, but rather a global setting I might be looking for.

billz, 2019-08-18:
> It worked indeed. To delegate wlan0 responsibilities to another adapter one has to:

This is clearly described in the FAQ.

closed 2019-08-18 by billz

pulkin, 2019-08-18:
Unless by FAQ you mean something else, [this](https://github.com/billz/raspap-webgui/wiki/FAQs#interfaces) is the only mention of `RASPI_WIFI_CLIENT_INTERFACE`. After reading it, I was even more convinced that `RASPI_WIFI_CLIENT_INTERFACE` has absolutely nothing to do with the access point interface. This phrase in particular
```
In this scenario, you may wish to use the built-in wlan0 interface with a USB wlan1 connection, rather than wired ethernet (eth0).
```
reads as "I want to use wlan1 instead of eth0". Well, I don't.

asyba, 2021-07-11:
@pulkin your" my external USB adapter " is another WIFI dongle or a usb3-etheener adapter?
Note: looking someway to use usb3-ethernet to join the AP without needed to use Wifi.