RaspAP/raspap-webgui #383: In AP mode, automatically connect to best gateway?

jzellis, 2019-08-18:
I'm not sure if I'm not understanding the AP setup properly, but here's what I'm trying to figure out.

I'm using a Pi as an AP/LAN router/Samba server in a travel trailer. Ultimately the trailer will be on the road and off-grid, and I want to set the Pi up so that I can use the Pi's AP to connect my various devices, like my phone, iPad, laptop and Chromecast to one another and the Pi's file server even if there's no WAN connection, so I can stream media locally, etc.

I've got raspap-webgui working and AP mode turned on and doing its thing, but it seems like I have to manually choose which hotspot it bridges to. What I'd like to do is add a list of SSIDs for it to connect to, preferably in order of priority, the way you can do it in wicd-curses, for example. The idea is that if I'm parked at home, it connects to my home wifi; if I'm out and about, it'll switch to using my phone's hotspot; if I'm near some public hotspot like a Starbucks, it'll connect to that.

Is this currently doable through the GUI? If not, can anybody point me to resources to set it up manually?

Slluxx, 2019-08-18:
you can do this in `/etc/wpa_supplicant/wpa_supplicant.conf`
if multiple are in range, the highest priority number will get picked.

```
network={
    ssid="HomeOneSSID"
    psk="passwordOne"
    priority=1
    id_str="homeOne"
}

network={
    ssid="HomeTwoSSID"
    psk="passwordTwo"
    priority=2
    id_str="homeTwo"
}
```

glaszig, 2019-08-19:
the problen is that any modifications to that config file will be overwritten by raspap.

i‘m working on a better implementation that‘ll merge existing config with raspap settings and show supported options configured manually in the ui.

i also have an idea for configuring priority. but that‘s in the future.

billz, 2019-08-19:
@jzellis excellent question. 

@Slluxx is correct. The `wpa_supplicant` utility automatically selects the best network based on the order	of the network blocks in the config file. A `priority` parameter may also be defined (defaults to 0). When multiple networks with the same priority are available, other factors like security policy and signal strength are used.

If networks are added in order of preference (default priority of '0'), `wpa_supplicant.conf` will be configured like so:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
	ssid="FreeWifi"
	psk=[passphrase]
}
network={
	ssid="Orange"
	key_mgmt=NONE
	scan_ssid=1
}
```

As @glaszig mentioned, this can be done via RaspAP's **Configure Wifi Client** UI. Manual edits will be squashed (for now).

jzellis, 2019-08-20:
Awesome, thanks! I knew about `wpa_supplicant.conf`, I just wasn't sure if there was some way to do it directly in raspAP. But this works perfectly for now.

closed 2019-08-20 by jzellis