RaspAP/raspap-webgui #386: Wifi sharing on 5GHz channel

kushvarma, 2019-08-19:
**Bug**
Everything works superbly untill I swtich my Wireless Mode from 802.11g to 802.11a to use 5 GHz Wifi Adapter.
If I keep everything by default and just change following things:
1. SSID
2. WPA to WPA2
3. Wifi password from ChangeMe to <something of my own>

Everything works as expected.
But when I go to Raspbian WiFi Configuration and in the Basic I switch from 2.4 GHz(g) to 5GHz(a) I cant connect to wifi anymore. The SSID is shown in the network available but after entering the password, it wont work. 

I am new to Raspberry Pi and still confused about a/b/ac/g 802.11 wifi adapters. 

**Your environment**
* Raspberry Pi 3 B+  
* Raspbian version: Buster Lite
* Followed the project prerequisites? Yes
* Checked the project FAQ? Yes
* RaspAP Quick Install or Manual setup? Quick
* Using default configuration? Yes
* Simultaneous AP and managed mode? No
* Onboard wireless chipset or external adapter? OnBoard 
* Other software or services running with RaspAP? No

**Steps to reproduce**
1. Use Quick installation to setup the raspberry pi.
2. It should be Raspbian Pi 3 B+
3. After connecting, go to 10.3.141.1 (default admin section)
4. In configuration Hotspot (http://10.3.141.1/index.php?page=hostapd_conf) go in the basic settings section.
5. Switch (801.11g - 2.4 GHz) to (801.11a - 5GHz)
6. Save settings and reboot from ssh
7. Try connecting to WiFi AP.

**Expected behavior**
The Wifi should be connected to the network via 5GHz channel.

**Actual behavior**
The SSID is shown in the network available but after entering the password, it wont work.

billz, 2019-08-19:
> I am new to Raspberry Pi and still confused about a/b/ac/g 802.11 wifi adapters

Did you set your WiFi country per the project prerequisites? 

Check the output of the following:

```
iw reg get
iw phy phy0 channels
```

A [gentle reminder](https://github.com/billz/raspap-webgui/wiki/Reporting-issues) that this is not a troubleshooting forum.

kushvarma, 2019-08-19:
I am sorry for creating a ticket, but I wasn't sure that is that a bug or an troubleshoot problem as I followed all the process as mentioned.
And to answer your question:
Yes I did, it is set to DE (Germany)

This is the output for `iw reg get`

```
pi@raspberrypi:~ $ iw reg get
global
country DE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```
and this is the output for `iw phy phy0 channels` is 
```
pi@raspberrypi:~ $ iw phy phy0 channels
Band 1:
	* 2412 MHz [1]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+
	* 2417 MHz [2]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+
	* 2422 MHz [3]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+
	* 2427 MHz [4]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+
	* 2432 MHz [5]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2437 MHz [6]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2442 MHz [7]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2447 MHz [8]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2452 MHz [9]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40- HT40+
	* 2457 MHz [10]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40-
	* 2462 MHz [11]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40-
	* 2467 MHz [12]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40-
	* 2472 MHz [13]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40-
	* 2484 MHz [14] (disabled)
Band 2:
	* 5170 MHz [34] (disabled)
	* 5180 MHz [36]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+
	* 5190 MHz [38] (disabled)
	* 5200 MHz [40]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40-
	* 5210 MHz [42] (disabled)
	* 5220 MHz [44]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+
	* 5230 MHz [46] (disabled)
	* 5240 MHz [48]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40-
	* 5260 MHz [52]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40+
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5280 MHz [56]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40-
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5300 MHz [60]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40+
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5320 MHz [64]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40-
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5500 MHz [100]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40+
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5520 MHz [104]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40-
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5540 MHz [108]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40+
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5560 MHz [112]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40-
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5580 MHz [116]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40+
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5600 MHz [120]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40-
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5620 MHz [124]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40+
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5640 MHz [128]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40-
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5660 MHz [132]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40+
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5680 MHz [136]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz HT40-
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5700 MHz [140]
	  Maximum TX power: 20.0 dBm
	  No IR
	  Radar detection
	  Channel widths: 20MHz
	  DFS state: usable (for 388 sec)
	  DFS CAC time: 60000 ms
	* 5720 MHz [144] (disabled)
	* 5745 MHz [149] (disabled)
	* 5765 MHz [153] (disabled)
	* 5785 MHz [157] (disabled)
	* 5805 MHz [161] (disabled)
	* 5825 MHz [165] (disabled)
```

I really appreciate any support on this.

Thank You.

glaszig, 2019-08-19:
i set the wireless mode to 5ghz which set `hw_mode=a` in `hostapd.conf` but i think it actually also needs `ieee80211n=1`. raspap only sets `ieee80211n=1` if `hw_mode=n`.

@kushvarma would you be able to test that combination manually?

kushvarma, 2019-08-19:
@glaszig thanks for replying. I tried setting `ieee80211n=1` in `/etc/hostapd/hostapd.conf` but it didn't work.
I reverted `hw_mode=g` and `ieee80211n=0` back now to run on 2.4 Ghz. Any other suggestions?

glaszig, 2019-08-19:
sorry, don‘t know how proficient you are but did you restart hostapd as well?

    sudo systemctl restart hostapd

kushvarma, 2019-08-19:
@glaszig yes I did.. It didn't work.. Also I tried restarting the pi after that, but did't work. don't worry on Linux side, I used to do DevOps for living(but ya never played with Pi and 802.11), throw me anything..

glaszig, 2019-08-19:
alright. weird. should work this way from [what i've read](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf).

maybe relevant: https://raspberrypi.stackexchange.com/questions/91074/how-to-configure-hostapd-for-a-5ghz-network

kushvarma, 2019-08-19:
@glaszig thank you. Let me try few more things. I have another pi 3b+ , let me try on that too. Just in case.
Danke

billz, 2019-09-06:
@kushvarma any updates? If not, will close this issue.

closed 2019-09-08 by billz