RaspAP/raspap-webgui #392: Multiples SSIDs

jeffersonpimenta, 2019-09-05:
I have several neighbors using my wifi and i want them conected in a separated ssid, no pass on it.

I want my devices conected under wpa, like i use now on OpenWRT.

Multiple ssid tool would solve my problem, can we implement it? 

I want also compatibility with pihole, its a awesome tool and can boost raspap-webgui use a lot.

Thanks a lot!

billz, 2019-09-06:
Lots of 'wants' in this issue ;) 

Re: multiple SSIDs, the capabilities reported by `iw list`:
```
valid interface combinations:
    * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
    total <= 4, #channels <= 1
```
clearly state that the 'number of simultaneous APs must be less than or equal to one'. In other words, you can't add a second interface of type `AP`. This is for the integrated wireless chipset on an RPi3b+. You are better off running a second AP on another RPi.

It _might_ be possible with an external adapter, but USB wifi dongles vary so much in capabilities that it would be practically impossible to support this.

> I want my devices conected under wpa

Check the **Security** tab of **Configure hotspot**.

> I want also compatibility with pihole

This is covered in the FAQ.

closed 2019-09-08 by billz

jeffersonpimenta, 2019-09-09:
My idea is use an external dongle with a better antena. 

I have some dongles laying around, like AWUS036NHA, my idea is use rpi like openwrt:

[image]

One radio tx two or more ssids, one have encription, another one has no encription.

Thanks.

glaszig, 2019-09-09:
> One radio tx two or more ssids, one have encription, another one has no encription.

like @billz wrote above: that capability depends on the wifi chip. the rpi chip supports max 2 ap‘s and is already being exhausted by raspap.

to have more ssid you‘ll need more external antennas/chips. that‘s generally possible but complicates setup and configuration and is currently not supported (in the ui).

billz, 2021-06-15:
A multiple AP setup with an external adapter is described here https://docs.raspap.com/multiple.html

ethical-haquer, 2025-03-30:
> A multiple AP setup with an external adapter is described here https://docs.raspap.com/multiple.html

Is there still a way to have multiple APs using an external adapter? That link now returns "404 - Not found". TIA.