RaspAP/raspap-webgui #393: Provide an API for mobile app to enable connect to wifi

thomas-hutterer-tik, 2019-09-08:
I am writing a mobile app that shall enable the user to configure the wifi settings of a raspberry pi, without forcing the user to interact with the raspberry pi directly. 

I want RaspAP to provide a REST formData HTTP endpoint that allows my app to provide the credentials for a wifi connection and switch the raspberry pi into connected mode with the new credentials.

I have been using comitup-web/connect API, but the comitup component has weak points that make RaspAP much more favourable.

I building a power measurement solution based on raspberry pi that should be very easy to install and use for non tech users right from the mobile app.

billz, 2019-10-04:
What can you contribute?

thomas-hutterer-tik, 2019-10-05:
I can contribute the API definition from comitup, which is open source and some sample code / unit tests how to use it to configure an IoT device from a mobile app

renerlemes, 2019-11-01:
Excelent idea. I'm also needing and looking for a solution like this for a home automation project I'm developing.

billz, 2019-11-08:
I could be wrong, but this issue feels a bit like offloading your work to us. If you'd like to take the lead on this, check out our [contribution process](https://github.com/billz/raspap-webgui#how-to-contribute), create a feature branch with the API framework you're interested in and open a PR. It needn't be code complete; just enough to provide a starting point.

We can then evaluate it and push commits to move it forward.

closed 2019-11-12 by billz