RaspAP/raspap-webgui #396: Access devices in the hotspot from the wired connection network

vaaski, 2019-09-23:
This might be an obvious thing, but I'm fairly new to this so please be understanding.
I have my main network - let's call it network A - which the raspberry pi is connected to (over eth0). Using raspap it's creating a WiFi hotspot (network B).
I have a device connected to network B, which I'd like to access from network A, but since the raspberry pi only has one local address in network A, I can only access the raspap webui.

Is there a way to do a kind of port-forwarding? E.g. from network A you'd navigate to http://raspberrypi:5345 which would load http://xyz:8080 from network B.

I don't know if that makes sense to you, if not please let me know.

tobimai, 2019-09-23:
Should be releated to #379, you basically want a bridge betwenn the ethernet and the WiFi instead of NAT.

billz, 2019-09-30:
Duplicate of #379

closed 2019-09-30 by billz