RaspAP/raspap-webgui #397: Accept defaults without prompt option for the quick installer

terence-bigtt, 2019-09-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. For example, I'm always frustrated 

Hi ! 
Actually, maybe it is possible already, but I didn't figure it out..
It would be nice to be able to run the quick installer with a '-y' like option, that accepts all default options.

Thanks !

billz, 2019-09-30:
You could pipe the results of the installer through the Linux `yes` command, like so:

`wget -q https://git.io/voEUQ -O /tmp/raspap && yes | bash /tmp/raspap` 

But I also think a command-line option `-y` or `--yes` similar to `apt-get` is a good suggestion for running installs non-interactively.

closed 2019-09-30 by billz

billz, 2019-09-30:
This has been implemented with the same approach as apt-get. So you can do an unattended install of RaspAP like so:

`wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap --yes`

The options `-y` or `--assume-yes` are also accepted and have the same result.

terence-bigtt, 2019-10-09:
Great ! Thanks a lot for the quick reply and dev !

commit 44f6762c (2019-09-30): Updated with unattended install option. Resolves #397