RaspAP/raspap-webgui #398: Use two wifi interfaces with predictable names.

keith24, 2019-09-23:
I was having trouble using the default wifi interface and a dongle as wlan0 and wlan1, because the interface names kept switching (sometimes the dongle was wlan0, sometimes it was wlan1).  I enabled predictive interface names in raspi-config, which fixed that.  

Now to use one interface as a client and one as hotspot, I followed [the instructions in the FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap).  

That worked, except that I couldn't scan wifi networks in the webgui.  

I fixed that by editing the sudoers file according to [#198](https://github.com/billz/raspap-webgui/issues/198#issuecomment-403086854).  I noticed the following lines already existed in the sudoers file: 

```
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan[0-9] scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan[0-9] scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan[0-9] reconfigure
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan[0-9] select_network
```

So all I had to do was add the same lines with the predictive interface name: 

```
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlx* scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlx* scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlx* reconfigure
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlx* select_network
```

So my feature request is... to include these lines by default in /etc/sudoers when raspap is installed.  That would allow predictive interface names to work out-of-the-box.  

"wlan[0-9]" could also be changed to "wl*".  Not sure if this is a security risk or anything.

taquitos, 2019-10-03:
@keith24, would you mind sharing how you configured things? I tried following the [same guide](https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap) as well as making the changes that you mention here with predictive interface names but I'm not able to get my wlan1(equivalent) to act as client. My wlan0 does broadcast but everything seems to route through the ethernet still. I'm trying to exactly what you seem to have accomplished 🙏

keith24, 2019-10-04:
All I did was what's in the FAQ, plus the edits to the sudoers file.

billz, 2019-11-06:
@keith24 thanks for posting your workaround. I'll do some testing around this and see if it can be incorporated into the project

Dude4Linux, 2020-01-22:
After spending an inordinate amount of time trying to get the scripts to work with predictable network names. Since the 'persistent network names' solution implemented in Jessie is no longer available, I came up with the following method of using traditional network interface names, eth0, wlan0 and wlan1.  Create a udev rules file as follows:
```
cat << EOT | sudo tee /etc/udev/rules.d/76-raspap-net-name.rules 
# Raspberry Pi Foundation Model <= 3B+
# Built-in Ethernet
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ENV{MATCHADDR}=="b8:27:eb:*", ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="eth*", NAME="eth0"
# Built-in WiFi
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ENV{MATCHADDR}=="b8:27:eb:*", ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="wlan*", NAME="wlan0"
# Raspberry Pi Trading Ltd. Model => 4B
# Built-in Ethernet
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ENV{MATCHADDR}=="dc:a6:32:*", ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="eth*", NAME="eth0"
# Built-in WiFi
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ENV{MATCHADDR}=="dc:a6:32:*", ATTR{dev_id}=="0x0", ATTR{type}=="1", KERNEL=="wlan*", NAME="wlan0"
EOT
```
Then reboot the Raspberry Pi.  The rules ensure that names are assigned to the built-in ethernet and wifi interfaces eth0 and wlan0 before names are assigned to the USB interfaces.  This has been tested on Models 3B and 3B+.  It should work on Model 4B's as well but hasn't been tested yet.

olewsaa, 2020-01-29:
I have put bindings into the interface file,  [/etc/network/interfaces](https://github.com/olewsaa/Yacht-computer/blob/master/wifi2wifi/web-version/interfaces) 
Something like this
auto wlan0
iface wlan0 inet manual
hwaddress ether b8:27:eb:7b:16:d0
auto eth0
iface eth0 inet manual
hwaddress ether b8:27:eb:2e:43:85

I left the wlan1 out as this is the only device left and I might change the long-range USB wifi antenna from time to time.

billz, 2020-03-12:
@Dude4Linux the best part of systemd's [docs on predictable network interface names](https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/) is "I don't like this, how do I disable this?" 😏 

The more I look at this, the more I'm convinced it's a terrible feature for the Pi

keith24, 2020-03-12:
> After spending an inordinate amount of time trying to get the scripts to work with predictable network names.

I'm totally lost here.  What scripts?  

I didn't have any trouble after changing my sudoers file.  No udev rules needed.  Are y'all talking about a different issue from the one I opened?

billz, 2020-03-13:
@keith24 predictable names are not the default on the Pi, but if these sudoers changes do no harm then I'm in favor of adding them to the project. I'll carve out some time and test it with the default config

closed 2020-05-15 by stale[bot]