RaspAP/raspap-webgui #399: raspap wifi scanning not working

humesrichard, 2019-09-28:
I installed raspap on rpi buster img, it worked after it was done installing and it scanned for wifi networks. for what i am doing i had to move it to a new folder called raspap so i can run mulit servers on it. everything works except for the wifi scanning part. im sure i broke something when i moved it to the folder just not sure where to look are what i can do to get it to scan again, any ideas?

billz, 2019-09-28:
Please check the [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#scanning).

humesrichard, 2019-09-28:
the scanning works when i no it in the terminal and i can scan networks using the rpi network stuff. just a pain to do that i want to go headless. do you have an idea what broke when in moved it in its own folder in the html folder so it wont scan on the web interface

humesrichard, 2019-09-28:
pi scans when using the terminal or using the pi native network app just will not scan on the web gui just wondering what i broke when i moved it in to its own folder

billz, 2019-09-30:
RaspAP in its current form needs to live in the web root. If you're running multiple sites, suggest you install these in subroots. An option may be added in the future to support a custom install location.

closed 2019-09-30 by billz

humesrichard, 2019-10-06:
i have tried to run subroots but failed. would apache2 have something to do with anything i have use apache for a certain server.

billz, 2019-10-06:
Best advice I can offer is to run RaspAP on a fresh install of Raspbian, following the project prerequisites. If you need to run a second web service, refer to [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#webport).