RaspAP/raspap-webgui #403: Allow for custom dnsmasq configuration via dnsmasq.d

coogle, 2019-10-01:
**Is your feature request related to a problem? Please describe.**

It would be very nice if by default the generated dnsmasq.conf file supported reading additional config files from a directory so that custom things not supported by the UI could still be done.

For example, in my case I want to route .local DNS queries to a different IP address for DNS lookup instead of a public DNS server so I need to add this to the config in addition:

```
server=/local/192.168.1.7
```

**Describe the solution you'd like**
Add the following to the auto-generated `dnsmasq.conf` so that `.conf` files can be stored in `/etc/dnsmasq.d` and left untouched by the UI

```
conf-dir=/etc/dnsmasq.d
```

I have done this to my dnsmasq.conf file, and basically just don't use the web UI for that at all anymore.

**Describe alternatives you've considered**
You could try to implement every single dnsmasq configuration setting, and have a whole new dnsmasq web-configuration-tool-project, but that seems unlikely!

billz, 2019-10-09:
Good suggestion and fairly simple to implement, thanks.

The project supports managing static DHCP leases with `dnsmasq.conf`, so I wouldn't rule out things like .local DNS queries in the future.

bmitchinson, 2020-02-25:
Is this ready to be picked up?

PR #501 (2020-02-26): add conf-dir option to dnsmasq.conf
this allows custom configuration stored in files in /etc/dnsmasq.d. raspap does not touch them and they'll always be loaded.

also creates /etc/dnsmasq.d if it doesn't exist.

closes #403.
commit 409adb0c (2020-02-26): add conf-dir option to dnsmasq.conf, closes #403

this allows custom configuration stored
in files in /etc/dnsmasq.d. raspap does not
touch them and they'll always be loaded.

also creates /etc/dnsmasq.d if it doesn't exist.
closed 2020-02-26 by billz