RaspAP/raspap-webgui #405: Can't access GUI!

formatc2013, 2019-10-04:
**Setup:**

Raspberry Pi Model B Rev 2

Raspbian GNU/Linux 8 Jessie

TP Link usb wifi dongle

I am about the set up my RP as a hotspot.
Used this command to install Raspap:

`wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap
`
It is working out of the box, I can connect to the network on my, smartphone etc, but I can't access the config on the RP. I had to change the hostapd config file to change ssid and passwd.

When I hover over the networking icon, it says:

wlan0 associated with raspi-webgui.

When I enter the ip 10.3.141.1 in the web browser I get a 500 error.

I can't find it in my network lists on the RP.

billz, 2019-10-04:
Did you follow the project prerequisites? Suggest you try this on a clean install. A 500 error indicates a conflicting web root and/or web service running on your RPi.

If your problem persists, use the provided issue template. It's there for a reason, thanks

closed 2019-10-04 by billz

iamogopogo, 2020-06-04:
I can confirm the error, as I've faced it numerous times, each and every time with a fresh install. It seems that after the initial setup in the browser, the login address changes to the IP address of the device. So, 10.3.141.1 became 192.168.0.174 for me, requiring that to log in again. I had this going on the 'old' Raspbian from a few months ago, but now with the newer Raspberry Pi OS, I can't seem to get the thing to broadcast. I can do it using network-manager-gnome and it's gui, but your program's the only way I could modify the channel width to get more than 72Mbps. I'm not smart enough to do it on my own, no matter how many guides I tried. :p

dwall60, 2020-06-25:
I have the same issue.
I can connect to raspi-webgui and access the internet. In fact, I'm using that right now to enter this info.
I can ping 10.3.141.1 successfully. But I can't get a response when putting the IP in my browser.
Chrome says ERR_CONNECTION_REFUSED.

Raspberry Pi 3
Raspbian GNU/Linux 8 Jessie

dwall60, 2020-07-02:
Moving to Buster fixed my issues.