RaspAP/raspap-webgui #406: Default hotspot configuration with WPA / TKIP is insecure

koenvervloesem, 2019-10-04:
**Describe the bug**
WPA / TKIP has been an insecure combination since years, for instance for [WPA packet spoofing and decryption](https://en.wikipedia.org/wiki/Wi-Fi_Protected_Access#WPA_packet_spoofing_and_decryption). Security researchers strongly recommend to switch to WPA2 / CCMP (AES).

**Your environment**
* Raspberry Pi 4 Model B
* Raspbian Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset
* Other software or services running with RaspAP? N

**Steps to reproduce**
Install RaspAP. Check hotspot security settings. See an insecure setup with WPA / TKIP.

**Expected behavior**
With RaspAP giving Wi-Fi clients access to my internal network, I'd expect the security to be better, so at least use a modern encryption protocol like WPA2 / CCMP.

**Actual behavior**
I observed the use of an insecure encryption protocol for the hotspot: WPA / TKIP.

**Additional context**
I know RaspAP wants to have a "known good" default configuration that "just works", but I find a "just works but insecure" configuration not acceptable for a system that gives attackers access to my network. I'd prefer to be secure, even if that means that a few people using older devices that only support WPA / TKIP can't connect to the RaspAP hotspot by default.

billz, 2019-10-04:
RaspAP does not create insecure networks, users of hostapd create insecure networks.

That is, RaspAP merely provides a management frontend for this service. TKIP is the default encryption protocol for hostapd. This project simply mirrors that setting.

koenvervloesem, 2019-10-04:
I understand that, but as a management frontend to make it easier to use services like hostapd, you should not blindly use the defaults of the services you use and you should consider more secure defaults. Because otherwise you put your users in danger. And someone will get hacked and will blame RaspAP, not an underlying service they have never heard about.

Security is a responsibility of everyone.

billz, 2019-10-04:
Per our [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#settings) we encourage users of RaspAP to develop an understanding of these services, rather than blindly clicking things. The RPi is, afterall, a learning platform.

koenvervloesem, 2019-10-04:
This really goes both ways: users have the responsibility to develop an understanding in the platform rather than blindly clicking things, while developers have the responsibility to not put their users at risk by blindly using defaults of services they use.

Please reconsider this. RaspAP is awesome, but this is not up to standard from a security point of view.

billz, 2019-10-09:
The same researchers who found the TKIP vulnerability have now also demonstrated that [WPA2/AES-CCMP is insecure](https://www.krackattacks.com/), so this not a panacea. 

Point is, nothing is really as secure as we think. No software is bulletproof and exploits exist in the wild that bypass zero-day vulnerabilities that we don’t yet know about.

> with RaspAP giving Wi-Fi clients access to my internal network, I'd expect the security to be better

RaspAP is heavily invested in practical security with many rounds of [audits and fixes](https://github.com/billz/raspap-webgui/pull/220) [[2]](https://github.com/billz/raspap-webgui/pull/357) by security-focused contributors, sophisticated CSRF token handling and a [guide on implementing HTTPS](https://github.com/billz/raspap-webgui/wiki/HTTPS--support), just for starters. We've also [recommended CCMP as a default setting change](https://github.com/billz/raspap-webgui/wiki/FAQs#wireless-mode) for some time now, but leave this up to the user.

As a developer, I consider the needs (and intelligence) of my user base, and balance this against the probability of an attacker using narrow and mostly theoretical exploits.

koenvervloesem, 2019-10-09:
I agree with your remarks about practical security and I appreciate your efforts to secure RaspAP, but let's just agree to disagree on this case of using more secure defaults for hostapd.

billz, 2019-11-09:
@CrypticOne thanks for the suggestion. I've been thinking along similar lines. The hostpad config is due for an update, and AC support is high on the list. Frankly, I've been putting it off because an AC AP with brcmfmac on a Pi 3b+/4 is not simply a matter of hardware capabilities. It must also take into account regulatory restrictions of the wireless spectrum. 

The regulatory info for brcmfmac is embedded in the firmware. Thus, an AC AP will fail to start if the country code is set to most European countries, for example (tested with several)

`nl80211: Failed to set channel (freq=5180): -22 (Invalid argument)`

Only combinations of certain frequencies (channels) and countries are capable of hosting an AC AP. The safest bet, probably, is to only enable it with:

```
channel=48 
country_code=US
```

I suppose this is the result when you ship one product with 5G stuff worldwide.

CrypticOne, 2019-11-12:
Thanks for the help getting it setup. I totally forgot about all the international issues with Wifi.  That adds a whole layer of issues.

billz, 2019-12-07:
Tracking the AC issue in #450

closed 2019-12-07 by billz