RaspAP/raspap-webgui #412: Client list table display incorrect

cwxyz007, 2019-10-13:
**Describe the bug**
Client list table display incorrect

**Your environment**
* Raspberry Pi hardware (examples: Pi 4)  
* Raspbian version (Buster Lite)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Y)
* Onboard wireless chipset or external adapter? Onboard wireless chipset
* Other software or services running with RaspAP? No

**Steps to reproduce**
1. Connect raspi-webgui with wifi
2. Login into dashboard with default username and password
3. Click ` Configure DHCP Server`
4. Click `Client list`

**Expected behavior**
Show the information of client in one row

**Actual behavior**
All client information show in one column

**Screenshots**
[image]

glaszig, 2019-10-13:
thanks for the report. that gets collaterally fixed when #410 will be merged.

billz, 2019-10-14:
Resolved by #410

closed 2019-10-14 by billz