RaspAP/raspap-webgui #413: Upgrade to SB Admin 2 v4 / Bootstrap v4

billz, 2019-10-15:
RaspAP is built on the legacy Bootstrap 3 version of [SB Admin 2](https://github.com/BlackrockDigital/startbootstrap-sb-admin-2), released in mid-2016. The [v4.0 release](https://github.com/BlackrockDigital/startbootstrap-sb-admin-2/releases/tag/v4.0.0) this year is a complete refactor based on Bootstrap 4 (itself a [major rewrite of the project](https://getbootstrap.com/docs/4.1/migration/#grid-system)). The layouts and Gulp build system have changed, Morris Charts dropped in favor of Chart.js, new pages and designs added, plus lots more.

Migrating RaspAP to the current releases of SB Admin 2 and Bootstrap 4 is a nontrivial task. But it's also an opportunity to update the project to the latest in responsive frameworks. @glaszig's work [rendering views from templates](https://github.com/billz/raspap-webgui/pull/384) streamlines this task in a big way. The most [recent release](https://github.com/billz/raspap-webgui/releases/tag/1.6.2) builds on his contribution to support this migration effort.

So how's it shaping up?

[image]

This is an evolution rather than a radical shift away from RaspAP's uncluttered design ethos. A few notable enhancements (so far, very much in progress):

1. Collapsible sidebar
2. Service/interface status indicators
3. Better all-around mobile device support
4. Font Awesome 5.0 icons
5. Chart.js based link quality meter
6. Stacking UI cards (panels are so yesterday) 
7. Improved UX

You can track the [feature branch](https://github.com/billz/raspap-webgui/tree/bootstrap4) in lead to a PR and eventual release. Feedback and beta testers are welcome.

wangkai88, 2019-10-16:
重启树莓派4b后wifi为撒老是变化呢

glaszig, 2019-10-18:
progress! i little too much red for my taste but i like 😉
still have a branch lingering with config parsers and don't get around integrating them 😢

billz, 2019-10-18:
@glaszig yeah, the 'raspberry red' was chosen somewhat arbitrarily for the first release :)

billz, 2019-10-20:
Migrated Morris Charts to Chart.js for data usage views.

[image]

closed 2019-11-01 by billz