RaspAP/raspap-webgui #425: Upgrading to version 2.0

feh123, 2019-11-02:
**Current Installation**
RPI +3B 
Buster Lite (latest update)
Compiled RTL8812AU for Linksys wifi dongle
Followed quick install of Raspap 1.6.2
Enabled wlan1 and managed mode.
End result - AP using wlan1 as hotspot that is working well.
**Issue**
Tried to upgrade to 2.0 using both FAQ methods: sudo git fetch --tags, sudo git checkout tags/2.0
and checkout master
Error message for both methods: your local changes to the following files would not be overwritten by checkout: includes/config.php
Please commit your changes or stash them before you switch branches.
Aborting.
Have tried rebooting and re-trying but get same error. 1.6.2 is still operational and Hotspot works.

billz, 2019-11-02:
No problem. Git is simply telling you that the changes to your local repo will be overwritten by the remote update. You can keep your changes to `includes/config.php` by stashing them. From the project root:

```
sudo git stash
git stash list
sudo git pull origin master
sudo git stash apply
```

feh123, 2019-11-02:
Thanks for fast response. When I first tried your commands I had the error that I was unknown - no email address or name. I added these and then ran your commands. I got the message Auto-merging includes/config.php                                                                        
On branch master                                                                                        
Your branch is up to date with 'origin/master'.                                                         
                                                                                                        
Changes not staged for commit:                                                                          
  (use "git add <file>..." to update what will be committed)                                            
  (use "git checkout -- <file>..." to discard changes in working directory)                             
                                                                                                        
        modified:   includes/config.php                                                                 
                                                                                                        
no changes added to commit (use "git add" and/or "git commit -a")
I rebooted and it has worked - looks good on my phone and tablet! Thanks!

closed 2019-11-02 by feh123