RaspAP/raspap-webgui #429: Can't use DNS server from wlan0 uplink in AP+Managed mode

sedgett, 2019-11-08:
**Describe the bug**

AP+Managed mode - unable to configure system to use the DNS servers provided via DHCP by the "uplink"  i.e. wlan0 client mode connection.

**Your environment**
* Pi Zero W
* Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ?  Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y/N) Y
* Simultaneous AP and managed mode? (Y/N) Y
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? lighthttpd updated to use ssl

**Steps to reproduce**
1.) Connected Managed mode interface to hotspot AP that happened to block access to Google's DNS 8.8.8.8.    
2.) Encountered no DNS resolution
3.) Verified connectivity to the internet worked via IP address but dns resolution failed.
4.) Confirmed /etc/resolv.conf showed 127.0.0.1 as name server.
5.) Confirmed dnsmasq was fwding to 8.8.8.8
6.) Commented out server=8.8.8.8 in dnsmasq and rebooted with same results.

**Expected behavior**
System to receive DNS server via client dhcp and update resolv.conf

**Actual behavior**
DNS queries sent to 8.8.8.8 (System set to 127.0.0.01 and dnsmasq set to forward to 8.8.8.8)

**Additional context**
Unsure how to even manually configure this.... and could entirely be a system level option that is not currently controlled by raspap. 

Additional troubleshooting - I tested by commenting out the server=8.8.8.8 option in /etc/dnsmasq.conf and deleting /etc/resolv.conf.  On reboot it was overwritten to 8.8.8.8.  Confirmed with a another client, that DHCP was not returning 8.8.8.8 as a dns server.  Read man pages for hostapd and dnsmasq but could not connect the dots to where the nuance is coming from.

As a future enhancement request, gui option to control in advanced tab would be an awesome feature.

billz, 2019-11-09:
@sedgett thanks for reporting, I'll check it out

colindawson, 2020-01-02:
I've just done a completely fresh install of raspbian buster full and have encountered the same problem.  DNS lookups on the PI are working fine, however anything connected to the hotspot fails.   I too would like the DNS Lookups to be forwarded on to the WIFI client connection.

specialcircumstances, 2020-01-07:
Not sure if this is related, but after reconfig or reboot, I appear to have to restart dnsmasq to get DNS resolution working.

colindawson, 2020-01-07:
Yes.    I just managed to get the dns lookup to work after restarting Dnsmasq.

So decided to try changing the pi Configuration to wait for network on boot and it worked straight off a reboot!

Problem solved.    I think that would be a useful addition to the faq.

Regards

Colin

billz, 2020-01-08:
@specialcircumstances @colindawson thanks for sharing your solutions. Generally speaking, it never hurts to give dnsmasq a kick. Agreed this would be a good candidate for the FAQ. I also think dnsmasq could use a logfile output option in the UI, similar to hostapd.

billz, 2020-01-10:
@sedgett does restarting dnsmasq resolve this for you?

specialcircumstances, 2020-01-11:
This appears to help, possibly resolve, and I've not seen any issues as a result (yet)

in /etc/dnsmasq.conf
```
#bind-interfaces                 # Bind to the interfaces
bind-dynamic                     # Use this method instead
```

man says:
```
       --bind-dynamic
Enable  a network mode which is a hybrid between --bind-interfaces and the default. Dnsmasq binds the address of individual interfaces, allowing multiple dnsmasq instances, but if new interfaces  or  addresses appear,  it  automatically listens on those (subject to any access-control configuration). This makes dynamically created interfaces work in the same way as the default. Implementing this option requires  non-standard  networking  APIs  and  it  is  only  available under Linux. On other platforms it falls-back to --bind-interfaces mode.
```

colindawson, 2020-01-11:
I like that answer.   It works for me as well.

Much better to be able to leave wait for network off.

It's also helping with the issue of switching between wifi networks.   
i.e. connecting the pi to my home network or my mobile hotspot.   works 
find if it can connect to a network on boot, but doesn't seem to like 
switching from one network to another on the fly.  (would be a great 
feature as it would make it run seamless)

Regards

Colin.

lostapathy, 2020-01-13:
I did a fresh install today and hit this issue.  Setting `bind-dynamic` in dnsmasq.conf seems to fix it for me, thanks!

billz, 2020-01-13:
Tested OK, now part of the default config. Thanks everyone for your input

billz, 2020-01-14:
Closing this. Feel free to reopen if needed.

closed 2020-01-14 by billz