RaspAP/raspap-webgui #432: [Question]how do you get so fast Access Point

yassinej, 2019-11-13:
Hello,

This is not a bug but a question regarding RaspAp implementation.

First of all, I have been using RaspAp for while now, and I have to say it is reliable, simple and performant!

Unfortunately, for a specific use-case, I can'y use RaspAp but I need to enable similar access point feature. So far, by mimicking RasAp I manage to get it working. I set up a 2.4Ghz AP and have internet access. 

My main problem is the speed.

I am using raspberry 3b+ and I can't get the same speed levels as when I am using RaspAp.
I have used same hostapd.conf file as the one generated by the RaspAp but it doesnt change.

Am I missing something? is there other settings need to have better speed?
It would be great to have some pointers where to have a look.

Thank you for your time.
Cheers.

billz, 2019-11-14:
All things being equal (hardware, driver, interface, hostapd binary, hostapd.conf, iptables, etc.) you should expect to see similar, if not identical, performance with these setups. There is nothing mysterious about RaspAP's [default config](https://github.com/billz/raspap-webgui/tree/master/config) or the manual installation steps. It's nice that you're getting better speed with RaspAP, but it's impossible to determine why with the information you've provided.

Please also be aware that this is not a troubleshooting forum, thanks. https://github.com/billz/raspap-webgui/wiki/Reporting-issues

yassinej, 2019-11-14:
Thank you for the feedback. I understand this is not a troubleshooting forum. I'll keep digging.

closed 2019-11-14 by yassinej

billz, 2019-11-14:
A couple of pointers:

1. if using the RaspAP-generated hostapd.conf, try shutting down the hostapd service and starting it manually with debug messages:
```
sudo systemctl stop hostapd.service
sudo hostapd -dd /etc/hostapd/hostapd.conf
```

2. You don't mention how you're comparing data transfer rates (ie., speed). A useful diagnostic tool for linux/raspbian is `speedometer`. After installing it, start the hostapd service and use this to measure and compare your data transfer rates:
```
sudo apt install speedometer
sudo systemctl start hostapd.service
speedometer -rx wlan0 -tx wlan0 -m $(( 1024 * 1024 * 3 / 2 ))
```
The latter will monitor traffic across `wlan0` and add labels to each peak in the graph.
Good luck and feel free to report your results.

yassinej, 2019-11-15:
Hi,
I figured out what was the issue. In my use case, I am using raspberry BLE capabilities.
since BLE is using the same 2.4Ghz frequency as Wifi, and Raspberry seems to have the same receivers for both. 

So I could achieve the same speed level as RaspAP when disabling Bluetooth service.

I understood later that I was aggressively scanning for devices. Changing the scanning interval, I could reduce interferences and hence improve wifi speed.

Cheers! and thanks for your pointers. I will start using the speedometer util!

billz, 2019-11-15:
Glad you isolated it! Thanks for sharing your results