RaspAP/raspap-webgui #434: Does not work in client mode

Cherkes001, 2019-11-18:
Hi, I used raspap about a month, until yesterday, I had to re-install the raspban system, then I started to install the necessary packages, installing raspap I was in trouble, it does not work in client mode. But as an access point, raspberries are sharing the network. I apologize if the bug report was already there, I did not find it.
Raspberry Pi 4B
Buster Lite
Quick Install

[image]
also in the screenshot you can see that there is some unknown network which has zero indicators, status, channel, etc.

billz, 2019-11-18:
Reinitialize wpa_supplicant per [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#wifi-scanning-doesnt-work-help) item.

billz, 2019-11-18:
This is not a bug with RaspAP, but it's possible the project can detect the error and reinitialize the daemon. In the meantime, you can debug and restart the daemon using the method above.

closed 2019-11-18 by billz