RaspAP/raspap-webgui #438: Doesn't work anymore on Raspian 8/PHP 5

maxgalbu, 2019-11-21:
The web interface doesn't work anymore on Raspbian 8 Jessie after this commit: https://github.com/billz/raspap-webgui/commit/7898dc24c84d72357391c3babe5ebe13e918e063

random_bytes() is available from php7 onwards. PHP 5 fails with error `(mod_fastcgi.c.2702) FastCGI-stderr: PHP Fatal error:  Call to undefined function random_bytes() in /var/www/html/includes/functions.php on line 63`

glaszig, 2019-11-22:
kind of intentionally, see https://github.com/billz/raspap-webgui/pull/356#issuecomment-518366811. php7 should be installed.

billz, 2019-11-22:
Support for Jessie / php5 is deprecated. The distribution itself is approaching EOL. 
This project will update to php7.3 for Raspbian 10 shortly.

maxgalbu, 2019-11-22:
I’ve done dist-upgrade but the OS remained on Raspbian 8. I will update to raspbian 9, no problem, but I think you need to disable installation altogether on 8 :)

commit a06de39d (2019-11-22): Deprecate Raspbian 8 / php5 support. Resolves #438
closed 2019-11-22 by billz