RaspAP/raspap-webgui #444: Missing commands in the manual instructions

jeffeb3, 2019-11-30:
I am integrating RaspAP into my pi image, which uses custompios. It basically mounts the file system in a qemu session emulating arm, and runs the commands from a script to build the image.

The first snag I hit was that the install script was trying to call restart on lighttpd, and it would have later done it again on hostapd. So since I'm making a script anyway, I decided to use your manual instructions.

There are two major parts missing (not including openvpn/mkcert stuff):

1) The config/ folder isn't being installed. These are the lines from the installer that I needed to replicate:

https://github.com/billz/raspap-webgui/blob/master/installers/common.sh#L231

2) The php optimizations weren't in the instructions either. I don't know anything about php, so maybe they aren't needed, but in case this is an oversight, I am pointing them out.

The copies from 1) should at least be in the readme. If you're hoping to get rid of the manual instructions, then maybe I could request adding an option for not calling `service`? Or is it just my weird environment that doesn't support it?

jeffeb3, 2019-11-30:
Also, the sudoers list is out of date. The list should match this:

https://github.com/billz/raspap-webgui/blob/master/installers/common.sh#L231

There are also 32 commands in that list, not 28:

https://github.com/billz/raspap-webgui/blob/master/installers/common.sh#L330

billz, 2019-11-30:
Thanks for pointing this out. Would you consider opening a PR?

jeffeb3, 2019-11-30:
Sure. Should the manual instructions include the php optimizations?

jeffeb3, 2019-11-30:
#445

billz, 2019-12-01:
@jeffeb3 much appreciated. The manual install steps are often the last to get updated. This is a big help, thanks for your contribution.

closed 2019-12-01 by billz

jeffeb3, 2019-12-01:
No problem. This is a pretty great tool. Keep it up.

FYI, here is the install script I'm using for custompios:
https://github.com/jeffeb3/v1pi/blob/master/src/modules/raspap/start_chroot_script

billz, 2019-12-01:
V1Engineering looks really interesting. Glad to see this project is useful as a component.