RaspAP/raspap-webgui #448: Using wlan0 and wlan1 cannot login into wifi

feh123, 2019-12-06:
Describe the bug

I am trying to run openvpn with wlan0 and wlan1 configuration. But I cannot get AP hotspot configured. I get the error message cannot execute wpa_cli reconfigure.

Your environment

    Pi +3 B
    Buster Lite
    Followed the project prerequisites? Y
    Checked the project FAQ? Y
    RaspAP Quick Install or Manual setup? Quick Install
    Using default configuration? (Y/N) Y
    Simultaneous AP and managed mode? (Y/N) Y
    Onboard wireless chipset or external adapter? Onboard for wlan0 external usb wifi for wlan1 Alfa dongle AWUS036NHA
    Other software or services running with RaspAP? None

Steps to reproduce
Repeated quick install with both wlan0 configured via wpa-supplicant or by using eth0. With eth0 I tried to configure wlan0 to onboard wifi but hit the above error message. With wlan0 and no etho (via adding wpa_supplicant file to boot) everything seems okay and I have a AP working - I have already edited config.php and dhcpcd.conf via FAQ. So the Alfa wifi is operational.
But on changing details of hotspot (name, security, AP mode etc) and doing a stop/start of the hotspot I lose wlan0. One trying to re-join wifi on wifi client page I get error message again re not able to execute wpa_client reconfigure. 

Expected behavior
AP working in managed mode with wlan0 and wlan1 - so far I have not tried to start openvpn.  
Actual behavior
I lose wlan0 and although I can see my client wifi on rescan I cannot join it.

billz, 2019-12-06:
Check this FAQ item
https://github.com/billz/raspap-webgui/wiki/FAQs#scanning

feh123, 2019-12-06:
Hi thanks for the fast response. I did try to re-initialise with this command but no success. I tried the commands from #368 which seemed to show wlan0 was configured but in use. 
I can get close to success with ssh and wpa-supplicant added to boot before starting powerup and using quick install. But restarting the hotspot seems fatal. You always show raspap pics with the wifi client in place. I have always used eth0 and do not seem to ge to a stable wifi client. 
Should I try going back to eth0 as the starting point?

billz, 2019-12-06:
Try it with eth0 first. I'm running OpenVPN with eth0 and the onboard wireless adapter on wlan0 with no issues.

> You always show raspap pics with the wifi client in place

this is just to demonstrate managed mode connectivity.

wlan0 and wlan1 configs are not officially supported by this project, simply because it's impossible to validate the capabilities of third-party wifi dongles. Not discouraging you from trying, just understand that I can't support you or assist with troubleshooting. There's an active thread on this topic which is probably a better place to discuss this.

feh123, 2019-12-06:
Thanks - I will start again with eth0 - I have it working with eth0 I just needed the portability of using wlan0/wlan1.

Which active thread are you thinking of? I will add any updates there. Thanks again.

closed 2019-12-06 by feh123

billz, 2019-12-06:
See https://github.com/billz/raspap-webgui/issues/335 Thanks :)