RaspAP/raspap-webgui #449: Wifi Client using (incorrect) 10.3.141.1 IP after rebooting

jordiblanchcarles, 2019-12-06:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Wifi is not using DHCP and is assigned 10.3.141.1 IP after connecting to a wifi access point, disabling "Wifi client AP mode" and restarting the Raspberry Pi.

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)  
Pi 3 Model B+, Pi 3 Model B and Pi 4 Model B+
* Raspbian version (examples: Buster Lite, Buster Desktop)
Buster Desktop
* Followed the project prerequisites? (Y/N)
Yes
* Checked the project FAQ? (Y/N)
Yes
* RaspAP Quick Install or Manual setup?
Quick Install
* Using default configuration? (Y/N)
Yes
* Simultaneous AP and managed mode? (Y/N)
Yes
* Onboard wireless chipset or external adapter? 
Onboard
* Other software or services running with RaspAP?
No

**Steps to reproduce**
I've installed latest Raspbian Buster, updated, upgraded and restarted the OS. Installed RaspAP using "curl -sL https://install.raspap.com | bash" and default options. Rebooted the RPi.
After rebooting, I can see the raspi-webgui wifi, connect to it and access the RaspAp web interface at 10.3.141.1 IP from another computer.
Now it is not possible to connect the RPi to any other wifi network because in "Configure Wifi client" section it is not detecting any wifi network.
Then I go to "Configure hotspot" section and under "Advanced" tab I enable "WiFi client AP mode" and save settings. Still under "Configure WiFi client" there's no way to see any wifi network.
I reboot again the RPi.
After reboot, the raspi-webgui wifi appears again and I can connect to it from another computer, but now the RPi IP has changed to 192.168.50.1 (instead of the original 10.3.141.1 IP). After connecting from another computer to the raspi-webgui wifi and accessing the new 192.168.50.1 from a browser, I can go to "Configure WiFi client" and then I can see the wifi networks available.
Then after add the passphrase to one of the networks, the RPi properly connects to that wifi network and the raspi-webgui AP is still present (always with 192.168.50.1 IP).
Now, as I don't want the RPi to work as an AP anymore (because I've already succeeded in connecting it to another wifi through the RaspAP web interface), I go to the web interface, "Configure hotspot -> Advanced" section, disable "Wifi client AP mode", save settings and stop the hotspot.
Right now everything seems correct, I've been able to connect to a wifi AP (my home router) and disable the RPi RaspAP hotspot, so my RPi is connected to my network and the hotspot is disabled.
But now if I reboot the RPi (or the power fails, or whatever happens that makes the RPi reboot) the RPi is no longer properly connecting to the configured wifi. If I take a look to the RPi using a keyboard and accessing with a terminal, I can see with "ifconfig" command that the wlan0 interface has taken the 10.3.141.1 IP and has not used the DHCP to get a correct IP from the router, but it shows that it is connected to the wifi I set up using the RaspAP web interface.
So, after properly configuring everything and then rebooting the RPi, it seems that RaspAP is meshing things and connects to the configured wifi router, but forces the 10.3.141.1 IP, so that the RPi is no longer accessible from the network.

**Expected behavior**
I'd expect that the system keeps working as configured, so after reboot it should connect to the configured wifi router and get a correct IP from DHCP server in the network and not force its 10.3.141.1 IP.

**Actual behavior**
After rebooting, the RPi connects to configured wifi router but forces 10.3.141.1 IP and doesn't use the DHCP server to get a proper IP in the network.

up2smoke, 2019-12-07:
+1 
1:1 the Same Problem.
Only what i have diffrent i have two wifi (wlan0 onboard, wlan1 plug-in)

jordiblanchcarles, 2019-12-12:
Hello everybody...

So nobody else is having this problem? Everyone is able to configure the wifi client using RaspAp when hotspot is not enabled?
I've tried several times, reinstalled Raspbian and RaspAp several times, used 2 different RPis and the problem is always there.

Am I doing something wrong?
Is this (wifi client not working unless hotspot is enabled) the intended default behaviour for RaspAP?

Please, I'd any kind of feedback because I do not know what else to try...

Thank you!

billz, 2019-12-13:
@jordiblanchcarles see a resolution in your subsequent (duplicate?) issue #452. 

Per the project readme, please be aware that you must have a wifi client configured _before_ enabling Wifi client AP mode. Toggling this on/off outside of managed mode will not achieve anything. This probably indicates a need to improve the UX with a disable state for this toggle.

billz, 2019-12-13:
Resolved with #455

closed 2019-12-13 by billz