RaspAP/raspap-webgui #450: Raspberry Pi 4 - Hotspot - 802.11ac

Thatoo, 2019-12-06:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Under Hotspot, option of wifi mode are limited to 802.11a,b,g,n. There is no 802.11ac option that the Raspberry Pi 4 can handle.

**Your environment**
* Raspberry Pi 4
* Raspbian version Buster Desktop
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install 
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset 
* Other software or services running with RaspAP? N

**Steps to reproduce**
Go to Hotspot and open the wifi mode option.
No 802.11ac option to select

**Expected behavior**
Get a 802.11ac option to be able to exploit fully Raspberry pi 4's capacity.

Thank you for your work

billz, 2019-12-07:
This is discussed here https://github.com/billz/raspap-webgui/issues/406#issuecomment-552117392

AC support is tricky due to regulatory restrictions embedded in the brcmfmac firmware. That said, it's an option I'd also like to see implemented.

Thatoo, 2019-12-07:
sorry, I had seen this subject bt couldn't understand it was talking about 802.11ac support. I'll follow this issue then.

I'm sure it is tricky and by opening this issue, my intention was not to put pressure on devs but just to contribute and track this issue.

billz, 2019-12-07:
@Thatoo no worries ;) that issue began with a discussion around default cipher protocols and morphed into AC support. Let's use your issue to track this, thanks

Thatoo, 2019-12-16:
How can I help you with this one?

Thatoo, 2019-12-20:
I have made a fresh install of buster on my raspi 4. I have installed Raspap.
I'm ready to help you trying and testing.
I'll have a look on internet to see if I can find tutorials explaining how to deal with 802.11ac on raspi 4.

Thatoo, 2019-12-20:
reading in here, https://github.com/raspberrypi/linux/issues/2619#issuecomment-410703338, I changed the original raspap's hostapd.conf file to the following one and I could get 802.11ac with up to 10MB/s speed!

```
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
auth_algs=1
beacon_int=100

ssid=raspi-webgui
wpa_passphrase=ChangeMe

country_code=US

interface=wlan0
driver=nl80211

wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

macaddr_acl=0

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=a
wmm_enabled=1

# N
ieee80211n=1
require_ht=1
ht_capab=[MAX-AMSDU-3839][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

# AC
ieee80211ac=1
require_vht=1
ieee80211d=0
ieee80211h=0
vht_capab=[MAX-AMSDU-3839][SHORT-GI-80]
vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42

ignore_broadcast_ssid=0
```

billz, 2019-12-21:
Glad to hear it. The key items here are the frequency (channel) and country_code. Per [this overview](https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_or_5.8_GHz_(802.11a/h/j/n/ac/ax)) of 5 GHz AC wifi channels, 36, 40, 44 and 48 are mostly unrestricted for indoor use. The AC option in RaspAP would need to populate these values in the channel select list.

For comparison, the wireless regulatory database used by Linux is [here](https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt).

Thatoo, 2019-12-22:
Now that I have solved my problem with the 802.11n (2.4Ghz) speed of raspi4 here, https://github.com/billz/raspap-webgui/issues/454 , If I can help you to solve the country-code issue with 802.11ac, let me know.

So far, no chance. If you have an idea I could try, let me know where to look at.

billz, 2019-12-25:
Testing RaspAP 802.11ac with iperf reports max throughput in the range of 75-100Mb/s 😎 

```
iperf -c raspberrypi.local
------------------------------------------------------------
Client connecting to raspberrypi.local, TCP port 5001
TCP window size:  129 KByte (default)
------------------------------------------------------------
[  5] local 10.3.141.197 port 63478 connected with 10.3.141.1 port 5001
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.0 sec   112 MBytes  94.2 Mbits/sec
```
Model: RPi 4 1GB / onboard wireless

billz, 2019-12-27:
Using [this source](https://w.wol.ph/2015/08/28/maximum-wifi-transmission-power-country/) with #465, I've built some test cases around countries which could, in theory, support 5 GHz  low-power ac channels that correspond to the RPi 4's hardware capabilities. Again, the relevant channels as reported by `iw list`:

```
Frequencies:
* 5180 MHz [36] (20.0 dBm)
* 5200 MHz [40] (20.0 dBm)
* 5220 MHz [44] (20.0 dBm)
* 5240 MHz [48] (20.0 dBm)
```
and _possible_ countries that permit these 100 mW / 20dBm channels:
```
$countries_5Ghz_max48ch = Array('AU', 'BR', 'CA', 'EE', 'HK', 'IL', 'IS', 'LT', 'LV',
'NZ', 'US', 'PH', 'SI', 'SK');
```
In practice, the only stable APs are obtained from "US". Monitoring dmesg and hostapd logs, the other country/channel combinations fail to start the AP for a variety of reasons, but mainly "could not set channel for kernel driver" (brcmfmac).

billz, 2019-12-27:
**Update:** setting the wifi country to "AU" with `raspi-config` _followed by a reboot_, stable APs are available on channels 36 & 44.

pjkemp, 2019-12-31:
where are with a release version that will support US and AU AP's and ac? this would be really cool!
thanks
PK

billz, 2020-01-01:
PR #465 is now merged with master. Refer to [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#80211ac) for more info. Thanks

roelfrenkema, 2020-07-01:
As per directions I added 

    "2_4GHz_max14ch": { 
        "countries": [ "NL" ],
        "channels": [ 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13 ]
    },

and

    "5Ghz_max48ch": {
        "countries": [ "NL" ],
        "channels": [ 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140 ]
    }

to wireless.jason and edited config.php. 

I can confirm a working AP for Netherlands up to channel 48, so the rest might as well be ommited.

Is there a way to add the other frequenties???

billz, 2020-07-04:
@roelfrenkema good to know, thanks. I'll update wireless.json accordingly.

Re: other frequencies, the 2.4 + 5GHz bands are defined in `includes/hostapd.php`. If you have an 802.11ax / wifi 6 adapter, for example, you're welcome to try it, although you will be limited by the RPi's bus speed.

victor-ab, 2020-09-15:
I'm getting:
```
HT40 secondary channel 44 not allowed
Fallback to 20 MHz
```
Any workaround for this? Country is set for US

MarinSNB, 2020-10-25:
I updated the wireless.json file and added there all the non-DFS channels up to 165, removed NL as one of the default countries under 5GHz section and left there US.  

I also edited the "includes/config.php" section and did the same thing (removed NL and left US there).  

Rebooted.  

All the 5GHz AC channels (that I had added) appeared in the GUI but would not save after selecting any of them.  For example, when I pick channel 161 or 165 and immediately click Save after that, the AC channel selection goes back to channel 36. I also made sure that US was selected in my Pi as well as the RaspAP GUI.

Any ideas on how to make any higher non-DFS channels (149, 151....up to 165) stick in the GUI?

Thank you!

billz, 2020-10-25:
@MarinSNB thanks for pointing this out. the hostapd channel validation  had a hardcoded upper limit of 48. It also didn't output a status message in the UI, which wasn't very helpful. I've fixed this and added a new constant `RASPI_5GHZ_MAX_CHANNEL` which defaults to 165.

MarinSNB, 2020-10-25:
Thank you so much!

billz, 2020-11-08:
@victor-ab see the HT capabilities section of [hostapd.conf](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf):

> Please note that 40 MHz channels may switch their primary and secondary
channels if needed or creation of 40 MHz channel maybe rejected based
on overlapping BSSes. These changes are done automatically when hostapd
is setting up the 40 MHz channel.

ie., if a band conflict is found hostapd will forcibly reduce the connection standard to HT20 and fallback to 20MHz.

pasman1, 2020-12-30:
Hi.
For country=US, i have stable connection on channels 153 and 161.
Raspberry Pi4.

Sujan-Roy, 2021-02-21:
I set up Raspberry pi 4 for 802.11ac as an access point. 
But I face one problem when changing the channel in #hostapd.conf file.
The problem is one channel (48)  throughput is high ( 78Mbps) and the next channel (52) throughput is low (40Mbps). 
Can you help me to fix this problem?
My configuration of hostapd.conf file is
 
interface=wlan0
driver=nl80211
logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
ssid=raspberrypi4
country_code=DFS-JP
ieee80211d=1
hw_mode=a
channel=40
beacon_int=100
dtim_period=2
max_num_sta=255
rts_threshold=-1
fragm_threshold=-1
preamble=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wmm_enabled=1
ieee80211n=1
ht_capab=[MAX-AMSDU-3839][HT40-][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
require_ht=1
ieee80211ac=1
vht_capab=[MAX-MPDU-3895][SHORT-GI-80][SU-BEAMFORMEE]
vht_oper_chwidth=0
eapol_key_index_workaround=0
eap_server=0
own_ip_addr=192.168.10.0
wpa=2
wpa_passphrase=raspberry
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

billz, 2021-02-21:
https://docs.raspap.com/

closed 2021-02-21 by billz