RaspAP/raspap-webgui #452: Wlan not working unless Hotspot is started with "WiFi client AP mode" enabled

jordiblanchcarles, 2019-12-09:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Unless the hotspot is enabled with "WiFi client AP mode" also enabled, the wifi client cannot connect to any network because it cannot see any network.

**Your environment**
* Raspberry Pi hardware
Pi 3 Model B
* Raspbian version
Buster Desktop
* Followed the project prerequisites?
Yes
* Checked the project FAQ?
Yes
* RaspAP Quick Install or Manual setup?
Quick Install
* Using default configuration?
Yes
* Simultaneous AP and managed mode?
Yes
* Onboard wireless chipset or external adapter?
Onboard
* Other software or services running with RaspAP?
No

**Steps to reproduce**
Install RaspAP on a clean Raspbian Buster.
Reboot.
Go to web interface on localhost -> Configure wifi client: There isn't any wifi SSID available.
Go to Configure hotspot -> Advanced: enable "WiFi client AP mode", save settings, stop hotspot and start hotspot again.
Go to Configure Wifi Client: press rescan button. Now we can see wifi networks. Add password to a wifi network and press add button.
Now RaspAP has connected the RPi to the wifi network and hotspot is also working. Everything is accessible, we can connecto to raspi-webgui hotspot and access the RPi, and we can also connect to the same wifi network as the RPi is connected and access the RPi. So everything seems ok.
Now, we do not need hotspot anymore (we have already connected the RPi to the wifi network), so we go to "Configure hotspot -> Advanced", disable "WiFi client AP mode", save settings and stop hotspot.
And here is where things break. After stopping hotspot, wlan0 still seems connected to the wifi network, but it's IP has changed to 10.3.141.1 and it is not possible to connect anywhere anymore. We cannot ping anywhere in our network or outide it (google.com for example) from the RPi although it shows that it is connected to the wifi network.
Rebooting the RPi doesn't change anything, and things are only working again if hotspot is restarted after enabling again the "WiFi client AP mode".

**Expected behavior**
Wlan should work if hotspot is not started and "WiFi client AP mode" not enabled.

**Actual behavior**
Wlan is not working unless Hotspot is started with "WiFi client AP mode" enabled.

jordiblanchcarles, 2019-12-11:
Hello everybody...

So nobody is having this problem? Everyone is able to configure the wifi client using RaspAp when hotspot is not enabled?
I've tried several times, reinstalled Raspbian and RaspAp several times, used 2 different RPis and the problem is always there.

Am I doing something wrong?
Is this (wifi client not working unless hotspot is enabled) the intended default behaviour for RaspAP?

Please, I'd any kind of feedback because I do not know what else to try...

Thank you!

feh123, 2019-12-11:
Hi

I do not use the wifi client configure page. My experience has been that the only successful way to operate RaspAp is to use eth0 and ignore the wifi client page. 
Whether I have configured the hotspot or not the wifi scan and connection is something I avoid because it never works well for me.
I am, by no reckoning, an expert on RaspAp (really a novice) so I have kept to a simple configuration. Then it works brilliantly and I greatly appreciate the originators's work! 
Whenever I have tried no eth0 I have configured a wifi client via a wpa_supplicant.conf file added to boot before startup and you can get to the web gui - as described in the installation instructions/FAQs. But if wlan0 is the wifi client then what broadcasts the AP? There is a thread on this re wlan0 and wlan1 configuration.
Not sure I have replied to your problem but I hope this helps.

jordiblanchcarles, 2019-12-12:
Hello @feh123,

thank you for your answer. I'm trying to use RaspAP in a new project where the RPis should be easily configurable by the users, that's why I wanted to use it and it's web interface, and this is also the reason not to use direct edition of wpa_supplicant.conf file or so...

> But if wlan0 is the wifi client then what broadcasts the AP?

That's why I try to use the "WiFi client AP mode" in hotspot configuration. In such configuration the same wlan0 can be used as a hotspot while being able to see/connect to other wifi access points. The problem is not this "WiFi client AP mode", this mode is working properly, but instead, after connecting to another wifi access point, disabling this mode and disabling the hotspot results in the wlan0 connecting to the other access point but getting an incorrect IP (it's forced to static 10.3.141.1 in dhcpdcp.conf file).

But anyway, thank you for answering.

Is there anyone out there that can tell me whether this behaviour is a bug or is an intended behaviour?

Thank you!

feh123, 2019-12-12:
Hi @jordiblanchcarles - thanks for the info re wlan0 and the AP - I was never sure what enabling it achieved because I had this idea that wlan0 for in and out would not be very speedy. If anything I considered enabling the AP was just a way to bridge to wlan1. I will try to see what speeds I can get. Good luck with your project.

billz, 2019-12-12:
@jordiblanchcarles this may come as a surprise, but I have a day job (and a life) outside of this project. RaspAP has some awesome contributors, but when it comes to building test cases, debugging and regressing issues, it's pretty much just me. So don't expect an immediate response.

Managed mode (or 'wifi client mode') has been largely trouble free since it was added. However, there have been some key updates to this part of the project. So it's possible this is a bug. I simply need to carve out some time to look at it.

Meanwhile, to anyone willing to pitch in, PRs are always welcome ;)

jordiblanchcarles, 2019-12-12:
Dear @billz ,

my apologies if I bothered you asking again for someone that could give me a clue whether this looked like a bug or it was just the intended behavior of RaspAP. I know most of us have a full time job (and life), so once again, please accept my apologies.

Meanwhile, I'll also try to search for the reconfiguration step that leads to this incorrect (at least from my point of view) wlan0 static ip after disabling hotspot.

I'll reply to this thread if I find anything.

Thank you!

jordiblanchcarles, 2019-12-12:
Hello again,

I think I've found where the issue comes from:

When disabling the "Wifi client AP mode" in "Configure hotspot->Advanced" and saving the settings, RaspAP is modifying the /etc/dhcpcd.conf file to a "kind of" default configuration, but I think this default configuration should consider whether there are already some wifi clients configured or not. If there is one or more wifi clients already configured, then RaspAP shouldn't add the "interface" lines at the end of the dhcpcd.conf file because this is causing the wlan interface not to get a correct IP configuration from dhcp server but instead force the static 10.3.141.1 IP.

I've been searching in the code where this is done and I've finally found that the file "includes/hostapd.php", line 247, checks whether wifi AP mode is enabled or not, and if it is not enabled, then it changes /etc/dhcpcd.conf file adding this default configuration that breaks any other wifi client configuration.

I've just commented out from line 269 to line 272 (both included) from "includes/hostapd.php" file and problem has gone away, now I'm able to disable the hotspot and reboot the RPi and wifi client is still working properly and I can connect to the RPi from the same wifi network.

I don't think just commenting out these lines should be the solution, because I think there would be other cases where these lines are useful, so I think that adding these lines should be done only when there are some wifi clients already configured. I'd suggest a patch for this if I knew how to do it, but I'm not used to php language nor RaspAP code, so I don't know how to implement this "if" conditional in a "RaspAP" way....

billz, 2019-12-13:
Breaking this down, I see two issues here. The first one:
> Install RaspAP on a clean Raspbian Buster. Reboot.
Go to web interface on localhost -> Configure wifi client: There isn't any wifi SSID available.

The **Configure wifi client** UI uses `wpa-cli` from wpa_supplicant to scan for available wifi networks. RaspAP does not supply a default wpa_supplicant.conf which, at a minimum, must include the following:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
```
Steps to initialize wpa_supplicant manually and test wpa-cli:
```
sudo nano /etc/wpa_supplicant/wpa_supplicant.conf      # add line above ^^ save & exit
sudo wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
sudo wpa_cli -i wlan0 scan
sudo wpa_cli -i wlan0 scan_results
```
This project has not supplied wpa_supplicant.conf as part of the default config, mainly to avoid stepping on user configs. There might be a case for installing it if not it's absent, just so this feature works "out of the box" without manual changes. I've tested the above on a clean install with Buster Lite and am able to scan and connect to wifi stations.

The second issue reads like a duplicate of your previous one #449. Is this a fair assessment? 

Finally, using Wifi client AP mode in the absence of a wireless client is not its intended use. This could be a UX issue, in which case the toggle should be disable when managed mode is not active.

billz, 2019-12-13:
> Is this (wifi client not working unless hotspot is enabled) the intended default behaviour for RaspAP?

Refer to the project readme and #455

closed 2019-12-13 by billz

jordiblanchcarles, 2019-12-14:
Hello again...

Sorry for bothering you again with this, but I've installed Raspbian Buster Desktop again and checked that after OS installation there is a /etc/wpa_supplicant/wpa_supplicant.conf file containing the "minimum" requirements. Also, just after OS installation
```
sudo wpa_cli -i wlan0 scan
sudo wpa_cli -i wlan0 scan_results
```
are also working and show a list of available SSIDs.

Then, after installing your latest RaspAP version and rebooting, these previous commands do not work (but I understand this is the intended behavior). But, now after your recent changes to code, I'm unable to connect the RPi to any wifi access point using the "Configure Wifi Client" in the RaspAP web gui. I've tried stoping the hotspot and rebooting, but there's no way to see other wifi networks in "Configure Wifi Client".

So is there any way to configure a wifi client using RaspAP web gui?

Thankyou!

billz, 2019-12-15:
I tested the latest installer on both Buster Desktop and Lite with a Pi 3 Model B. I am able to scan wifi networks and configure a wireless client with RaspAP. Be sure to follow the project prerequisites. I am unable to assist you in troubleshooting further. Best of luck.

jordiblanchcarles, 2019-12-16:
Ok, thank you for your time.

mlavelle42, 2020-01-15:
I have the same problem, same situation (Buster...). Were you able to resolve it?

jordiblanchcarles, 2020-01-15:
Hello @mlavelle42 ,

my solution was to clone a previous version of the RaspAP repository and make the [changes](https://github.com/billz/raspap-webgui/issues/452#issuecomment-565101558) that worked for me. You can find it [here](https://github.com/jordiblanchcarles/raspap-webgui).

You can also use the RaspAP Quick Installer just changing the repo: 

`curl -sL https://install.raspap.com | bash -s -- --repo jordiblanchcarles/raspap-webgui`

Please note that I also made a modification that prevents the authentication from showing when the RaspAP WebGUI is opened from localhost (it was a requirement in my project).

With this modification of RaspAP I'm able to configure the Raspberry Pi to start as an AP, enable "WIFI cllent AP mode", search/connect to new wifi networks and then disable AP mode while wifi is still connected to new network without loosing the IP it got from DHCP server.

Hope it helps!

mlavelle42, 2020-01-18:
Thought I had it (well, almost). No luck really. Are you running Stretch or Buster?

jordiblanchcarles, 2020-01-23:
Hello @mlavelle42 ,

I'm using Buster, this is what I do to make the RPi work as in AP mode and still be able to connect to another WiFi network (similar to what a brand new WiFi repeater would do...):

at pi shell install my version of RaspAP:

`$ curl -sL https://install.raspap.com | bash -s -- -y --repo jordiblanchcarles/raspap-webgui`
`$ sudo reboot`

after these steps you should have a new wifi named 'raspi-webgui'.

Connect to the 'raspi-webgui' wifi using password 'ChangeMe'.
Go to '10.3.141.1' using your web browser. Insert admin password when prompted (admin - secret). The RaspAP web gui should appear.
Go to "Configure hotspot - > Advanced". Enable "Wifi client AP mode" and save settings.
Reboot again the RPi.
Now you should see once again the 'raspi-webgui' wifi network, but if you connect to it, the new RPi network address will be '192.168.50.1' (I don't know why there's a different configuration of the IP in AP mode when "Wifi client AP mode" is enabled...).
Connect once again to the 'raspi-webgui' wifi network and go to '192.168.50.1' IP using your web browser.
Under "Configure WiFi client" you should be able to see other wifi networks and you should also be able to connect to them while RPi is still in AP mode.
To disable the AP mode and keep the RPi connected to your wifi network just go to "Configure hotspot -> Advanced", disable "Wifi cliennt AP mode", "Save settings" and "Stop hotspot".

Hope it helps!!

commit 78b09cc4 (2019-12-13): Initialize wpa_supplicant, create if absent. Resolves #452