RaspAP/raspap-webgui #453: Can't connect to AP

brad, 2019-12-10:
**Description**
Can't connect to the AP after a fresh install. The AP is there, but clients say the password (ChangeMe) is invalid

**My environment**
* Raspberry Pi 1 Model B
* Raspbian Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install, default options but no OpenVPN
* Using default configuration? Y
* Simultaneous AP and managed mode? I don't know the answer to this
* Onboard wireless chipset or external adapter? Edimax WiFi dongle
* Other software or services running with RaspAP? Clean raspbian install

**Steps to reproduce**
Install using the quick installer on a fresh Raspbian Buster install on a Raspberry Pi 1 Model B

**Expected behavior**
I expect to be able to connect to the `raspap-webgui` network

**Actual behavior**
I cannot connect to the `raspap-webgui` network. The wifi client tells me the password is invalid

**Related StackOverlow Issues**
https://raspberrypi.stackexchange.com/questions/77178/setting-up-raspap

billz, 2019-12-10:
Have you tried it with a newer Pi that has onboard wifi?
This is not a troubleshooting forum for USB wifi dongles.

brad, 2019-12-10:
Oh, I didn't mean to use this as troubleshooting. From my reading of the docs, this should be a supported situation:
- I couldn't find a list of supported raspberry pi devices so I assumed they are all supported
- The wifi dongle I'm using is the one recommended [in the docs](https://github.com/billz/raspap-webgui#prerequisites)

I will run hostapd with debugging and see if anything jumps out in the logs

billz, 2019-12-10:
Recommended does not mean supported. The Edimax EW-7811Un has worked reliably with the driver/firmware and hostapd included with Raspbian in the past, however this may no longer be the case. This project does not test, validate or support any external wifi adapters.

brad, 2019-12-10:
Fair enough, sorry for wasting your time

closed 2019-12-10 by brad

brad, 2019-12-12:
I think the problem I was having was #2979
I ended up installing OpenWRT instead, where this dongle is working well