RaspAP/raspap-webgui #454: Raspberry pi 4 - wifi drop the speed

Thatoo, 2019-12-11:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
When I connect by ssh to my Raspberry Pi 4 and I download a file, I can get speed up to 11MB/s which is awesome (Raspi 3 could only get 5-6MB/s because of CPU limitation using OpenVPN).
However, if I try downloading the same file from my computer connected by wifi to my Raspap (Raspi 4 or 3 it does the same), the speed drops to 2.5MB/s .

**Your environment**
* Raspberry Pi 4 
* Raspbian version Buster Desktop
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset
* Other software or services running with RaspAP? a static website (HTML/CSS) that's all

**Steps to reproduce**
ssh pi@10.3.141.1
wget http://proof.ovh.net/files/1Gio.dat

download with high speed

same download but from a computer connected to wifi to raspap
wget http://proof.ovh.net/files/1Gio.dat

download at 2.5MB/s

**Expected behavior**
maybe not a download as high as the raspberry can (10-11MB/s) but something higher than this 2.5MB/s...

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

Thatoo, 2019-12-11:
I hope I can help to solve this issue

Thatoo, 2019-12-11:
Connected to my raspap (raspberry pi 4) with /etc/hosts pointing to the local ip of my raspap, I have tried to download a file from it to my computer (directly by wifi, peer to peer, no internet, no vpn...) and the result of the download is 2.7MB/s. Is it really the wifi speed of a raspi 4 using raspap or am I getting strange result?

vutranHS, 2019-12-13:
I have experienced the same. My wired connection got download speed around 80 Mbps. When i connect to Rasp wifi hotspot the speed only exceed at 20 up to 25 Mbps. Any ideas ?

billz, 2019-12-13:
@Thatoo duplicate of your previous issue #450? You fail to mention several things, principly the wireless mode, which makes diagnosing this difficult if not impossible.

vutranHS, 2019-12-13:
> @Thatoo duplicate of your previous issue #450? You fail to mention several things, principly the wireless mode, which makes diagnosing this difficult if not impossible.

We use the default settings: Wireless Mode (tried 802.11a - 5 GHz, 802.11b - 2.4 GHz, 802.11g - 2.4 GHz, 802.11n - 2.4 GHz), CHannel (tried 1, 6, 7, 11) but the hotspot wifi speed drops to around 20 - 25 Mbps all the time.

billz, 2019-12-13:
Assuming you are initially connecting via ssh with wpa_supplicant, there is obviously a difference with local throughput versus what a remote client will observe with hostapd. Is this is speed comparison you're trying to make?

billz, 2019-12-13:
> I have experienced the same. My wired connection got download speed around 80 Mbps. When i connect to Rasp wifi hotspot the speed only exceed at 20 up to 25 Mbps. Any ideas ?

Even if your Pi4 is sitting on top of a router, ethernet will be faster than wifi. Unless you can clarify this issue, I'll mark it as closed.

billz, 2019-12-13:
Re-reading OP's report, it's clear this is a speed comparison of local wireless throughput vs a client connected via hostapd. This is not a valid comparison, nor is it a bug with this project.

closed 2019-12-13 by billz

Thatoo, 2019-12-14:
I'm sorry, I don't understand everything you say and I believe you didn't get what I said.. Let me try with a drawing.

Thatoo, 2019-12-14:
My VPN provider offers a speed of 11-12 MB/s (88-100Mbps).

With the raspi 3B (802.11n) on which I installed Yunohost and this two apps : 
- https://github.com/labriqueinternet/vpnclient_ynh
- https://github.com/labriqueinternet/hotspot_ynh

If I connect to ssh to my raspi 3 and I make my raspi 3 download  `wget http://proof.ovh.net/files/1Gio.dat` , I get speed at about 5-6 MB/s
I stop ssh.
From my computer, connected to internet by wifi to my raspi3 (with VPNclient ON) I try to do the same, `wget http://proof.ovh.net/files/1Gio.dat`, the max speed expected is obviously this 5-6MB/s (the CPU can't handle the speed of 11-12MB/s that my VPN provider offer) but I get only half of it, 2.5MB/s....
So to see where the problem comes from, I'm trying to download a file from my raspi 3 localy over wifi to test the wifi of raspi 3 and I can download at 5-6 MB/s.
So the raspi 3 not only cannot deal with the full speed of VPN (encryption is too difficult process for its CPU that can offer only 5-6 MB/s) but it also can't provide this speed over wifi (even though the wifi speed couldallow that) because (I have been explained that) the bus of wifi is shared or something...  

Now with the raspi 4 (802.11n selected in raspap, yunohost is not yet available on raspi 4)

If I connect to ssh to my raspi 4 and I make my raspi 4 download  `wget http://proof.ovh.net/files/1Gio.dat` , I get speed at about 11 MB/s which is awesome and prove that the raspi 4's CPU can handle encryption much better.
I stop ssh.
From my computer, connected to internet by wifi to my raspi4 (with Raspap and VPNclient ON) I try to do the same, `wget http://proof.ovh.net/files/1Gio.dat`, the max speed expected is obviously this 11 MB/s but I get also only 2.5MB/s.... which is not half of it this time but one fourth (really sad)...
So to see where the problem comes from, I'm trying to download a file from my raspi 4 localy over wifi to test the wifi of raspi 4 (as I did with raspi 3) and I was expected at least 5-6 MB/s (like for raspi 3, hoping for more) but I can download only at 2.5 MB/s.
So I don't know if the problem comes, as it is apparently the case for raspi 3, the wifi bus being shared or something, because apparently the wifi is limited to that speed of 2.5MB/s.

With 802.11n (2.4Ghz), I was expected to get a download similar to raspi 3 like 5-6 MB/s and I was hoping that later, when 802.11ac will be available will even gain more but for now, downloading a local fil (saved inside the Raspi 4) over wifi (local download), I get speed of only 2.5MB/s and that is what I'm trying to understand why is that.

Is it possible to optimize something in my raspi 4 to be able to download over wifi local files at speed of 5-6 MB/s?

Thatoo, 2019-12-14:
`wget http://10.3.141.1/1Gio.dat` is downloaded from my raspi 4 (with raspap) at only 2.5 MB/s whereas from my raspi 3 (with above mentioned apps) it is downloaded at 5-6MB/s.

Thatoo, 2019-12-14:
> Re-reading OP's report, it's clear this is a speed comparison of local wireless throughput vs a client connected via hostapd. This is not a valid comparison, nor is it a bug with this project.

I'm not sure to understand that but I'm now comparing the wifi speed on local network of raspi 3 and 4 as hotspot.

Thatoo, 2019-12-14:
I'm looking around and I have noticed weird stuff :
I read the README. The file /etc/sudoers is empty. Shouldn't it contain a long list?
```
www-data ALL=(ALL) NOPASSWD:/sbin/ifdown
www-data ALL=(ALL) NOPASSWD:/sbin/ifup
www-data ALL=(ALL) NOPASSWD:/bin/cat /etc/wpa_supplicant/wpa_supplicant.conf
```
and so on?

Then an other weird stuff, if now I do `sudo raspi-config` , Localisation options, Change Wi-fi country, it tells me 

> Could not communicate with wpa_supplicant

> There was an error running option I4 Change Wi-fi Country

Maybe that helps you to see where is my problem. I don't understand because I did it once following the readme Prerequisites and Quick installer

billz, 2019-12-15:
I am unable to assist you with troubleshooting. We have your other open issue to track AC support for hostapd. I will address this when I find time, thanks.

Thatoo, 2019-12-16:
I understand.

Thatoo, 2019-12-20:
I have reinstall a fresh install of Buster on raspi 4, following step by step (not installing any VPN this time) and reach to the same situation where the wifi speed is limited to 2.5MB/s.

Let's focus on 802.11ac and hope it will get better then.

Thatoo, 2019-12-22:
I have solved my issue. Below is the hostapd.conf  that worked for me.
Based on my previous post that allowed me to make 802.11ac work with country-code =US, I have played with it to make 802.11n working at full speed of Raspi4 : https://github.com/billz/raspap-webgui/issues/450#issuecomment-568125435

Now, with the following hostapd.conf file, I can respect rules set by my country (different than US), and I can use 802.11n (with 2.4Ghz) at speed of 12MB/s (about 100Mbps, known as a high speed for raspberrys). I hope it will help others and @billz to improve his awesome work. Thank you bilz for your work.

```
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
auth_algs=1
beacon_int=100

ssid=raspi-webgui
wpa_passphrase=ChangeMe

country_code=FR

interface=wlan0
driver=nl80211

wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

macaddr_acl=0

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=g
wmm_enabled=1

# N
ieee80211d=1
ieee80211n=1
require_ht=1
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40][MAX-AMSDU-3839]
channel=1

ignore_broadcast_ssid=0

```

Thatoo, 2019-12-22:
Can be closed!

Thatoo, 2019-12-27:
I haven't touch to my raspi 4 since then except a reboot and I have lost the full speed of 802.11n...
I don't get it.
And I got back this weird error message when I want to do sudo raspi-config , Localisation options, Change Wi-fi country, it tells me

```
    Could not communicate with wpa_supplicant

    There was an error running option I4 Change Wi-fi Country
```
any idea @billz ?

billz, 2019-12-27:
Here is the fix https://github.com/billz/raspap-webgui/wiki/FAQs#scanning

Thatoo, 2019-12-28:
This indeed helped me to solve the `sudo raspi-config` (Localisation options/Change Wi-fi country) problem. and I can also, as expected and explained in the page you gave me, get result from `sudo wpa_cli -i wlan0 scan_results`
However, it last only till a reboot... after a reboot, the issue is there again and I again need to do
`sudo wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0 `
and then, I need to restart hostapd to get my wifi back with `sudo systemctl restart hostapd.service`
I can't reboot or I'll get back the issue with wpa_supplicant.

All that is not big deal as I we are not supposed to reboot our raspi very often.

I'm sad because I can't get back the speed I once had, 12MB/s (100Mbps). I got only half of it, 6MB/s which is better that what I had at the very beginning of this thread (2.5MB/s) but still, it's annoying not to be able to get the maximum of 802.11n of raspi 4.

Any idea where to look at?
I'll try with country=US just to see what I get.

Thatoo, 2019-12-28:
Whatever modification I make, I never get the 150Mbps back, I only get 72Mbps... country being FR or US, ieee80211d being 1 or 0, channel being 1 or 3 or 6 or 8... Sadly I get only 72Mbps not 150Mbps...
any idea @billz

billz, 2019-12-29:
In order to achieve anything close to 150 Mbps, the wireless stream must operate at a 40 MHz wide channel on the 2.4 GHz band. A 20 MHz channel will restrict you to 72 Mbps. Your hostapd.conf has the required settings, but this is no guarantee of a 40 MHz channel.

In practice, this can be quite difficult due to interference on the 2.4 GHz band. There are many things that will cause an AP to fallback to 20 MHz. The most common reason is if an AP detects another wireless network within 40 MHz, i.e. two channels, of its own channel. For example, if an AP is set to channel 6, another network operating anywhere from channel 4 to 8 will trigger a fallback.

Check out [this resource](https://www.lmi.net/wp-content/uploads/Optimizing_802.11n.pdf) on optimizing 802.11n.

Generally speaking, you will have a better chance of achieving the speeds you're after on the 5 GHz band.

Thatoo, 2020-01-01:
I'm learning a lot with your. Thank you!

billz, 2020-01-01:
@Thatoo it's a great question. I'll add it to the FAQ.