RaspAP/raspap-webgui #455: Managed Mode(AP and Client) not working properly

kernsche, 2019-12-13:
* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
If the RPI will bootup with the raspap.service, the "AP" Network disapears after the dhcpcd.service gettings started by servicestart.sh script.

In /var/log/syslog, hostapd will log:

Dec 13 01:14:04 raspit hostapd[624]: uap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Dec 13 01:19:50 raspit hostapd[547]: Configuration file: /etc/hostapd/hostapd.conf
Dec 13 01:19:50 raspit hostapd[547]: Could not read interface uap0 flags: No such device
Dec 13 01:19:50 raspit hostapd[547]: nl80211: Driver does not support authentication/association or connect commands
Dec 13 01:19:50 raspit hostapd[547]: nl80211: deinit ifname=uap0 disabled_11b_rates=0
Dec 13 01:19:50 raspit hostapd[547]: Could not read interface uap0 flags: No such device
Dec 13 01:19:50 raspit hostapd[547]: nl80211 driver initialization failed.
Dec 13 01:19:50 raspit hostapd[547]: uap0: interface state UNINITIALIZED->DISABLED
Dec 13 01:19:50 raspit hostapd[547]: uap0: AP-DISABLED
Dec 13 01:19:50 raspit hostapd[547]: uap0: CTRL-EVENT-TERMINATING
Dec 13 01:19:50 raspit hostapd[547]: hostapd_free_hapd_data: Interface uap0 wasn't started
Dec 13 01:19:50 raspit systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Dec 13 01:19:50 raspit systemd[1]: hostapd.service: Failed with result 'exit-code'.
Dec 13 01:19:52 raspit hostapd[632]: Configuration file: /etc/hostapd/hostapd.conf
Dec 13 01:19:52 raspit hostapd[632]: uap0: interface state UNINITIALIZED->COUNTRY_UPDATE

If i comment out `systemctl start dnsmasq.service`, the hostapd seems to work. When i "manual" doing wpa_supplicant the same behavior will occur as with dhcpcd.service.

If i will connect the RPI to a Wireless Network as a Client with the Webinterface, everything will be fine and the AP-Mode(hotspot) will work.

**Your environment**
* Raspberry Pi 4
* Buster Lite
* Followed the project prerequisites? (Y/
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset
* Other software or services running with RaspAP? No

**Steps to reproduce**
Not Connect to a existing Wifi Network.
Configure raspap as "WiFi client AP mode", reboot and see if there is a AP Network.

**Expected behavior**
Getting an AP Network without connection to an existing Network as Client but with the option to connect to a Network.

**Actual behavior**

[image]

[image]

[image]

[image]

**Additional context**

[image]
[image]

[image]

[image]

Sometimes everything will work as expected, but most of the time its not working.

billz, 2019-12-13:
> Not Connect to a existing Wifi Network.
Configure raspap as "WiFi client AP mode", reboot and see if there is a AP Network

Are trying to use Wifi client AP mode without a wifi client connection?

billz, 2019-12-13:
Related to #449 #452. Seems users are confused about this feature, or assume that enabling it somehow creates a wifi client configuration.

closed 2019-12-13 by billz

SilverWx, 2020-07-08:
I've experimented the same problem after setting up headless raspbian, if I enable Wifi client AP mode without chaging AP nsettings, it will work, but if I change one AP settings, for exemple ssid, the uap0 interface disapear.

SilverWx, 2020-07-08:
Ok, I find my problem,
First time I setup WiFi client AP mode it works, hostapd.ini file is ok, but after, each time you save your hoastapd config, hostapd.ini will be reset to original state instead of 👍 `WifiInterface = upa0
LogEnable = 1
WifiAPEnable = 1
BridgedEnable = 0
WifiManaged = wlan0
`

commit 80133ece (2019-12-13): Disable toggle if managed mode not enabled. Resolves #455