RaspAP/raspap-webgui #457: DHCP Server not working in managed mode

stebla27, 2019-12-17:
* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
I have configured RASP AP in managed mode.
Before everything was working fine.
But now no DHCP server is running on the Wifi server side anymore and my clients get no valid IP.
Changing the DHCP server settings does not help.

**Your environment**
* Raspberry Pi 4
* Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Managed mode; using AP and Client
* Onboard wireless chipset or external adapter? Onboard
* Other software or services running with RaspAP? No

**Steps to reproduce**
Enable managed mode and connect via wifi by a wifi client.

**Expected behavior**
Wifi client gets valid IP because DHCP is running.

**Actual behavior**
No IP

billz, 2020-01-06:
I'm able to reproduce this. Introduced with #329. Fix is forthcoming.

billz, 2020-01-06:
@stebla27 thanks for reporting.

closed 2020-01-06 by billz

billz, 2020-01-06:
The issue here is toggling default config states (managed AP vs. unmanaged) while also reading values from the system dnsmasq.conf. You could prevent this by re-configuring DHCP with the UI between these managed/unmanaged AP toggle states, but I suspect most users want the convenience of having these default values handled for them.

commit ff2c91fe (2020-01-06): Bugfix, resolves #457