RaspAP/raspap-webgui #458: Feature request -- official docker container

bd8392, 2019-12-18:
Hi :-) It would be really nice to see an official docker-contianer for this software :-). Because I use to run software ONLY in container like rancher os (keeps mantainance low). Any possibility to see this? :-) maybe I can help developing :-)

mlavelle42, 2020-01-24:
Second this request.

billz, 2020-01-27:
@bd8392 I don't really have bandwidth for this now, but if you'd like to kickstart things perhaps myself and others could collaborate on it :)

glaszig, 2020-03-01:
enlighten me, please. how does docker handle management of the host’s network interfaces? built-in and usb dongles? can your run hostapd inside a container?

glaszig, 2020-03-01:
sorry, i enlightened myself. apparently you can indeed. looks like quite some work: https://github.com/fgg89/docker-ap

bd8392, 2020-03-01:
By bind the etc/interfaces to docker as volume, you can alter it's settings...maybe this is easier...

glaszig, 2020-03-01:
- https://github.com/v1s1t0r1sh3r3/airgeddon/wiki/Docker%20Linux
- https://github.com/tsaikd/docker-builder/tree/master/net-wireless/hostapd/latest
- https://github.com/brannondorsey/mitm-router

glaszig, 2020-03-01:
> By bind the etc/interfaces to docker as volume, you can alter it's settings...maybe this is easier...

go, go, go!

noxPHX, 2020-03-10:
I managed to run RaspAP within a Docker container without it being privileged or running within the host network namespace (the wifi interface must be moved into the container netns so to speak) for some security concerns. 
I still stumble upon some web interface errors but nothing serious as the AP is working great so far so I need to do some more tweaks as well as checking the capabilities needed before sharing my work to you if you are interested.

bd8392, 2020-03-11:
Wow that's great! Yes I'm always interested in dockerized apps :-) thanks for all the effort you've done! @noxPHX. Maybe you can post it here, so the docker can become official and others can use it :-). Very cool that you've done it :-)

danielwilson654, 2020-05-25:
Great stuff! I heavily use docker so because of  #579 a direct install of raspAP won't work in my situation. 

@noxPHX Any progress on the container? if not perhaps share the docker image so we can work on it together?

jrcichra, 2020-06-11:
Sure @billz that works for me. 

However, we could also investigate what breaks RaspAP when docker is installed on the same system and how we can fix it? The install process for RaspAP is pretty streamlined as it is. A docker container might be overkill for how many systemd services are involved. It may turn into a docker-compose deployment, behind more networking complexity.

I think the investigation approach also solves @danielwilson654 's problem. We just want to run Docker and RaspAP on the same Pi.

billz, 2020-06-11:
@jrcichra I'm running docker v19.03 with the latest release of RaspAP on RPi OS Lite; no obvious issues. What sorts of problems are others seeing?

[This repo](https://github.com/garyritchie/docker-raspap) is quite old, but maybe it provides a starting point?

jrcichra, 2020-06-12:
I'm setting up a Pi4 fresh with RPi OS Lite. I installed docker with `sudo apt install docker.io`, which is `18.09.1+dfsg1-7.1+rpi1+rpt1`. I'll try installing from docker directly.

jrcichra, 2020-06-12:
Did a fresh install of RPi OS Lite, did `sudo apt update && sudo apt upgrade`, then `curl -fsSL get.docker.com -o get-docker.sh && sh get-docker.sh` as shown on https://www.docker.com/blog/happy-pi-day-docker-raspberry-pi/

Then I installed RaspAP with `curl -sL https://install.raspap.com | bash`.

I changed the SSID to `J71012` and set a different password. This is what my phone tells me:

[image]

billz, 2020-06-12:
I installed docker using the same method but did so _after_ installing RaspAP.

jrcichra, 2020-06-12:
Yep, installing RaspAP, _then_ docker, it works in NAT or in bridged mode. That's fascinating!

noxPHX, 2020-06-12:
Hey guys, sorry I have been really busy lately but I am of course interested to continue this project.
I still have my Dockerfile but it might be a bit outdated now... I will try to rebuild it asap.
I have my own Docker architecture and my image depends on it but I can create a public repo for it and I will share it if that sounds good to you ?
I will get in touch as soon as possible

jrcichra, 2020-06-12:
Sounds great. I'd love to see what you've done.

bd8392, 2020-06-12:
Wow very cool to see this evolving :-)

billz, 2020-06-12:
I've created [this public repo](https://github.com/billz/raspap-docker) which could serve as a community-led project. Or we can use @noxPHX's repo as a starting point, whichever is easiest.

jrcichra, 2020-06-25:
The first challenge I'm seeing is `systemd`. I'm trying a very simple Dockerfile:

```
FROM balenalib/raspberrypi3:buster
RUN install_packages wget curl systemd
RUN curl -sL https://install.raspap.com | bash
```

To see if we can reuse what already is built. It defaults to Yes on all options but does not play nicely with `systemd`:

```
System has not been booted with systemd as init system (PID 1). Can't operate.
```

There are several daemons RaspAP relies on and that may throw off the "single process per container" model.

rugk, 2020-09-18:
@jrcichra Well, then you maybe need to use a systemd-enabled Docker container. See https://github.com/alehaa/docker-debian-systemd

jrcichra, 2020-10-03:
@rugk Thanks for the systemd container. I was able to build a simple Dockerfile and get the web GUI up and running. See my PR: https://github.com/billz/raspap-docker/pull/1

systemd needs to be the initial process in that container. Users will have to start the container detached, then run the setup script inside, which uses the cURL call.

billz, 2020-10-05:
@jrcichra PR merged, thanks. the limited testing I've done is to access the GUI from the container. it's a great PoC starting point.

rusagent, 2021-02-03:
How can i pull and run the arm64 docker image? With the supplied commands it always pulls the amd64 version :/

billz, 2021-02-03:
@rusagent try downgrading your Docker version. see https://github.com/pi-hole/docker-pi-hole/issues/735

hwacookie, 2021-02-28:
> How can i pull and run the arm64 docker image? With the supplied commands it always pulls the amd64 version :/

Or, instead of downgrading docker, you could also just clone https://github.com/RaspAP/raspap-docker (thank you, @billz !) and create your own docker image with a simple "docker build ."

1ubuntuuser, 2021-04-29:
@noxPHX What settings did you use for the non host network version? I'm setting this up so it will with the docker-compose and docker swarm.

noxPHX, 2021-04-29:
Hello and sorry for missing for such a long time, 
I've just pushed to https://github.com/RaspAP/raspap-docker on the develop branch my updated work for the 2.6.5 release.
At the moment the AP is working great but I've not implemented additional features such as the VPN or the AdBlock. The web interface does not work as intended as well, I didn't had time to fully investigate but I feel like this issue is impacting me somehow (https://github.com/RaspAP/raspap-webgui/issues/803).
I will try to work on it but this project seems now more mature and complete than the first days I was trying to build my container, this will add some complexity but I'm glad RaspAP is evolving this way.
Feel free to help me!

1ubuntuuser, 2021-05-09:
@noxPHX thanks I've had a look. I'll get testing. Does client mode work with your setup?

luandro, 2021-11-01:
@jrcichra I see u were trying to get RaspAP to work with Balena. Have you managed to get it deployed?

I started a discussion around this at the [Balena forum](https://forums.balena.io/t/acessing-sys-fs-cgroup/349846/7). But still have no idea how to move forward.

billz, 2021-11-01:
@luandro we'd love to see RaspAP deployed with Balena.
frankly @jrcichra is the real docker expert here. I'm available to help push things forward

jrcichra, 2021-11-02:
Hi, I have a spare Pi I can qualify with the Balena docker image. I have more container/Linux know-how since last year so I should be able to assist more. Things like adblock and VPN should work if we set `--network=host`. Interacting with hardware will be done with the `--privileged` flag. Essentially we'll be using the docker container to isolate all the config files and processes, but with none of the networking abstractions.

We'll want to make sure nothing on the host system will clash with what's being described in the RaspAP container. Disabling daemons like `wpa-supplicant` outside the container might be necessary. It might take some work to get it feature-compatible...and it might take more than just a `docker run` for the end-user.

luandro, 2021-11-02:
This is the Balena docker-compose I'm trying:
```
version: '2.1'

services:
  hotspot:
    image: jrcichra/raspap-docker:latest  
    privileged: true
    network_mode: host
    labels:
      io.balena.features.sysfs: '1'
    cap_add:
      - SYS_ADMIN
```

jrcichra, 2021-11-02:
Ok. I think I'm where you're at. The guide at https://github.com/RaspAP/raspap-docker still works where the webgui comes up. But starting the hotspot in the container fails:

```
Nov 02 21:10:02 jpi systemd[1]: hostapd.service: Service RestartSec=2s expired, scheduling restart.
Nov 02 21:10:02 jpi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 8.
Nov 02 21:10:02 jpi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Nov 02 21:10:02 jpi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
Nov 02 21:10:02 jpi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Nov 02 21:10:02 jpi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Nov 02 21:10:02 jpi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
```

...but that's a separate problem we can get to once we're on Balena.

I remember now that the `jrcichra/raspapd` is based on some random systemd container. A potential first step is getting systemd running in Balena. The downside of systemd in a container is...when you're building the container, systemd is not running. So the install script needs to do `systemctl enable <service>` and not `systemctl enable --now <service>`. That's why in the existing container, I have the user do the install script by hand.

So step 1 seems to be Balena systemd to support `hostapd`, `dnsmasq`, `dhcpcd`, `lighttpd`, etc easily. https://github.com/alehaa/docker-debian-systemd might be a good reference we can slide Balena under.

jrcichra, 2021-11-02:
I was able to get systemd at least running in the container with this `Dockerfile` slimmed down from the work https://github.com/alehaa/docker-debian-systemd did: 
```
FROM balenalib/raspberry-pi-debian:buster
RUN install_packages systemd systemd-sysv cron anacron
RUN systemctl mask --   \
    dev-hugepages.mount \
    sys-fs-fuse-connections.mount
RUN rm -f           \
    /etc/machine-id \
    /var/lib/dbus/machine-id
ENV container docker
STOPSIGNAL SIGRTMIN+3
VOLUME [ "/sys/fs/cgroup", "/run", "/run/lock", "/tmp" ]
CMD [ "/sbin/init" ]

```

jrcichra, 2021-11-02:
I'm where I was at before. Using Belena as a base is much nicer, especially with `install_packages`. I'm getting this output for `hostapd` when enabling the hotspot:
```
Nov 02 22:20:21 jpi systemd[1]: Stopped Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Nov 02 22:20:21 jpi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator...
Nov 02 22:20:21 jpi hostapd[562]: Configuration file: /etc/hostapd/hostapd.conf
Nov 02 22:20:21 jpi hostapd[562]: rfkill: WLAN soft blocked
Nov 02 22:20:21 jpi hostapd[562]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Nov 02 22:20:21 jpi hostapd[562]: wlan0: Could not connect to kernel driver
Nov 02 22:20:21 jpi hostapd[562]: Using interface wlan0 with hwaddr b8:27:eb:97:25:f1 and ssid "raspi-webgui"
Nov 02 22:20:21 jpi hostapd[562]: Failed to set beacon parameters
Nov 02 22:20:21 jpi hostapd[562]: wlan0: Could not connect to kernel driver
Nov 02 22:20:21 jpi hostapd[562]: Interface initialization failed
Nov 02 22:20:21 jpi hostapd[562]: wlan0: interface state COUNTRY_UPDATE->DISABLED
Nov 02 22:20:21 jpi hostapd[562]: wlan0: AP-DISABLED
Nov 02 22:20:21 jpi hostapd[562]: wlan0: Unable to setup interface.
Nov 02 22:20:21 jpi hostapd[562]: wlan0: interface state DISABLED->DISABLED
Nov 02 22:20:21 jpi hostapd[562]: wlan0: AP-DISABLED
Nov 02 22:20:21 jpi hostapd[562]: wlan0: CTRL-EVENT-TERMINATING
Nov 02 22:20:21 jpi hostapd[562]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Nov 02 22:20:21 jpi hostapd[562]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Nov 02 22:20:21 jpi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Nov 02 22:20:21 jpi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Nov 02 22:20:21 jpi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
```

It looks like we can't interact with the wlan0 kernel driver:
```
Nov 02 22:20:21 jpi hostapd[562]: wlan0: Could not connect to kernel driver
```

...even though we should be able to interact with the host kernel (in theory).

I got here by doing:
```
docker build -t jrcichra/balena-systemd .
docker run --name raspap -it -d --privileged --network=host -v /sys/fs/cgroup:/sys/fs/cgroup:ro --cap-add SYS_ADMIN jrcichra/balena-systemd
docker exec -it raspap bash
curl -sL https://install.raspap.com | bash
... go through the prompts saying yes
let it restart the container
docker restart raspap
```

Then the webgui should appear on port 80 and if you do
```
journalctl -u hostapd -f
```
...while trying to turn on RaspAP you should be in the same boat with the wlan0 failure.

jrcichra, 2021-11-02:
Ok, by dealing with `rfkill` and disabling `systemd-resolved` where I could, I was able to start up `hostapd` and `dnsmasq` in the container. I can see the wifi SSID and the password works. Where I'm stuck now is with `dnsmasq-dhcp`. I can't get an IP address from the AP. I'm getting this error, but the config (I believe) is set up properly to serve dhcp:

```
==> 090_wlan0.conf <==
# RaspAP wlan0 configuration
interface=wlan0
dhcp-range=10.3.141.50,10.3.141.255,0.0.0.0,12h
dhcp-option=6,1.1.1.1,1.0.0.1

==> README <==
# All files in this directory will be read by dnsmasq as 
# configuration files, except if their names end in 
# ".dpkg-dist",".dpkg-old" or ".dpkg-new"
#
# This can be changed by editing /etc/default/dnsmasq

root@jpi:/etc/dnsmasq.d# tail -f /tmp/dnsmasq.log
Nov  2 23:10:27 dnsmasq[1145]: using nameserver 10.0.0.1#53
Nov  2 23:10:27 dnsmasq[1145]: read /etc/hosts - 5 addresses
Nov  2 23:10:29 dnsmasq[1145]: read /etc/raspap/adblock/hostnames.txt - 142490 addresses
Nov  2 23:10:35 dnsmasq-dhcp[1145]: no address range available for DHCP request via wlan0
Nov  2 23:10:39 dnsmasq-dhcp[1145]: no address range available for DHCP request via wlan0
Nov  2 23:10:44 dnsmasq-dhcp[1145]: no address range available for DHCP request via wlan0
Nov  2 23:10:49 dnsmasq-dhcp[1145]: no address range available for DHCP request via wlan0
Nov  2 23:10:53 dnsmasq-dhcp[1145]: no address range available for DHCP request via wlan0
Nov  2 23:11:02 dnsmasq-dhcp[1145]: no address range available for DHCP request via wlan0
Nov  2 23:11:16 dnsmasq-dhcp[1145]: no address range available for DHCP request via wlan0

root@jpi:/etc/dnsmasq.d# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:c2:70:a4 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.182/24 brd 10.0.0.255 scope global dynamic noprefixroute eth0
       valid_lft 85808sec preferred_lft 75008sec
    inet6 fd38:65c3:b703:0:2080:69f:2cd0:851/64 scope global mngtmpaddr noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fdb3:dd5c:79f7:0:e0d2:632b:a43b:2b51/64 scope global mngtmpaddr noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::c106:e0a:badc:12c9/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:97:25:f1 brd ff:ff:ff:ff:ff:ff
    inet 169.254.151.177/16 brd 169.254.255.255 scope global noprefixroute wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::f46f:c75e:5fbd:ad5b/64 scope link 
       valid_lft forever preferred_lft forever
4: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:01:9f:bf:c4 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
root@jpi:/etc/dnsmasq.d# 

```

luandro, 2021-11-02:
@jrcichra Not sure it helps, but this is how I usually setup dnsmasq/dhcp:
```
interface=wlan0 # Listening interface
dhcp-range=192.168.4.2,192.168.4.20,255.255.255.0,24h
                # Pool of IP addresses served via DHCP
domain=wlan     # Local wireless DNS domain
```

jrcichra, 2021-11-02:
@luandro yeah, I'd think that would work but for some reason it doesn't:
```
root@jpi:/etc/dnsmasq.d# head -999 *
==> 090_adblock.conf <==
conf-file=/etc/raspap/adblock/domains.txt
addn-hosts=/etc/raspap/adblock/hostnames.txt

==> 090_raspap.conf <==
# RaspAP default config
log-facility=/tmp/dnsmasq.log
conf-dir=/etc/dnsmasq.d
log-dhcp
log-queries

==> 090_wlan0.conf <==
interface=wlan0 # Listening interface
dhcp-range=192.168.4.2,192.168.4.20,255.255.255.0,24h
                # Pool of IP addresses served via DHCP
domain=wlan     # Local wireless DNS domain

==> README <==
# All files in this directory will be read by dnsmasq as 
# configuration files, except if their names end in 
# ".dpkg-dist",".dpkg-old" or ".dpkg-new"
#
# This can be changed by editing /etc/default/dnsmasq

systemctl restart dnsmasq

root@jpi:/etc/dnsmasq.d# tail -f /tmp/dnsmasq.log
Nov  2 23:31:54 dnsmasq-dhcp[1244]: DHCP, IP range 192.168.4.2 -- 192.168.4.20, lease time 1d
Nov  2 23:31:54 dnsmasq[1244]: reading /etc/resolv.conf
Nov  2 23:31:54 dnsmasq[1244]: using nameserver 10.0.0.1#53
Nov  2 23:31:54 dnsmasq[1244]: read /etc/hosts - 5 addresses
Nov  2 23:31:56 dnsmasq[1244]: read /etc/raspap/adblock/hostnames.txt - 142490 addresses
Nov  2 23:32:16 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
Nov  2 23:32:18 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
Nov  2 23:32:22 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
Nov  2 23:32:31 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
Nov  2 23:32:48 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
Nov  2 23:32:56 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
Nov  2 23:33:01 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
Nov  2 23:33:05 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
Nov  2 23:33:14 dnsmasq-dhcp[1244]: no address range available for DHCP request via wlan0
^C
root@jpi:/etc/dnsmasq.d# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:c2:70:a4 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.182/24 brd 10.0.0.255 scope global dynamic noprefixroute eth0
       valid_lft 84455sec preferred_lft 73655sec
    inet6 fd51:de9e:9c6d:0:2404:4818:84c2:454a/64 scope global mngtmpaddr noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fd32:db85:3f4f:0:db2a:61b0:9b30:ebe3/64 scope global mngtmpaddr noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fd38:65c3:b703:0:2080:69f:2cd0:851/64 scope global mngtmpaddr noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fdb3:dd5c:79f7:0:e0d2:632b:a43b:2b51/64 scope global mngtmpaddr noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::c106:e0a:badc:12c9/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:97:25:f1 brd ff:ff:ff:ff:ff:ff
    inet 169.254.151.177/16 brd 169.254.255.255 scope global noprefixroute wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::f46f:c75e:5fbd:ad5b/64 scope link 
       valid_lft forever preferred_lft forever
4: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:01:9f:bf:c4 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
root@jpi:/etc/dnsmasq.d# 

```

NL-TCH, 2024-05-09:
@billz @jrcichra 
this one can be closed right?
as https://github.com/RaspAP/raspap-docker is stable (https://docs.raspap.com/docker/)

billz, 2024-05-10:
@NL-TCH it sure can, thanks

closed 2024-05-10 by billz