RaspAP/raspap-webgui #459: Managed mode not wavailable on Pi zero icm buster 10.2

LeaTaka, 2019-12-18:
**Before submitting an issue**
* [x] Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
On my PI zero W I did: sudo apt-get dist-upgrade, now on buster version 10.2 / Linux 4.19.75+
The result is that I cannot enable managed mode (WiFi client AP mode).
How can I help troubleshooting this?

**Your environment**
* PI Zero W
* Buster Lite 10.2
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install & Manual setup
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Y)
* Onboard wireless chipset 
* Other software or services running with RaspAP? (N)

**Steps to reproduce**
sudo apt-get dist-upgrade
sudo reboot
install raspap
sudo reboot
I reach the raspap webserver through usb0 successfully

**Expected behavior**
Enable managed mode as this worked on version 1.6 icm with Stretch

**Actual behavior**
Button is not available

billz, 2019-12-18:
Some users ran into problems by activating a managed mode AP without first configuring a wireless client. [This commit](https://github.com/billz/raspap-webgui/commit/5462e26d9520085a420f8fe897ad0d9e39e8c323) added detection for the managed mode state, and uses this to enable the feature in the UI.

 If you're connecting to the Pi via usb OTG, I assume you don't (yet) have a wireless client config, thus the 'Wifi client AP' button is not enabled. Is that correct?

If so, can you describe how you used this feature in v1.6? Thanks

LeaTaka, 2019-12-18:
That's correct, however just now connected through wifi and got the same result (in stretch and buster).

Correction: there is a high possibility that I used raspap version 2.0 successfully before. So I did install that previous raspap version (2.0?) through OTG. Then before boot I enabled managed mode and rebooted. This worked.

If desired I can try to reproduce this with version 2.0 on Buster and Stretch as well.

billz, 2019-12-18:
> If desired I can try to reproduce this with version 2.0 on Buster and Stretch as well

That would be helpful indeed, thanks.

LeaTaka, 2019-12-18:
- Raspbian GNU/Linux 9.11 (stretch after dist-upgrade)
- RaspAP 2.0

_Configure hotspot findings:_
Advanced tab: WiFi client AP mode was allready enabled.
Basic and Security tab: settings were not preset.
After saving new settings a complaint that saving was not possible, however settings were saved.
Cannot save the interface preset though, it will be preselected by raspap
Start hotspot fails
In the right top of the page it says: hostapd down

_Configure wifi client findings:_
A nice overview of all AP's
I could connect to a client without a problem
internet is available on the rpi0

_Dashboard findings:_
Interface Name wlan0
In the right top of the page it says: uap0 up

LeaTaka, 2019-12-18:
**- Raspbian GNU/Linux 10 (buster after dist-upgrade)**
**- RaspAP 2.0**

_Configure hotspot findings:_

1. Advanced tab: WiFi client AP mode can be enabled.
2. Basic and Security tab: settings were not preset.
3. After saving new settings a complaint: Unable to save wifi hotspot settings.
4. Only interface setting was not saved, it defaults to uap0.
5. No RaspAP SSID in my PC list though
6. Stop and start hotspot succeeds
7. In the right top of the page it says: hostapd up
8. The the SSID shows in my PC AP-list, I can connect
9. Then I rebooted, moved away from OTR and tried to use the wireless client and failed: no SSID available in my lPC list
10. going back through OTR and restarting the hotspot from webgui makes the SSID available in my PC list again
11. However a reboot makes the SSID not come back in the PC list

I also checked sudo systemctl restart raspap.service and did a requested systemctl daemon-reload, however no real difference.

_Configure wifi client findings:_

1. A nice overview of all AP's
2. I could connect to a client without a problem
3. internet is available on the rpi0

_Dashboard findings:_

1. Interface Name wlan0
2. In the right top of the page it says: uap0 down
3. I can strat it without a problem, it says uap0 up
4. after reboot: in the right top of the page it says: uap0 down
5. I can strat it without a problem, it says uap0 up

LeaTaka, 2019-12-18:
**My usecase:** Visitors need to be able to connect their rpi0 to an AP that changes each day. They can power the rpi0, connect through their smartphone and use RaspAP to connect to the rpi0 to the AP with their daily chnaging credentails.

billz, 2019-12-19:
@LeaTaka thanks for sharing these detailed findings. I'm traveling at the moment but will have time to investigate once I'm settled.

LeaTaka, 2019-12-20:
**Some interestung findings for 2.1 on Buster 10;**

1. Booted pi zero w
2. Connected through raspi-webgui
3. Could not see wifi clients, only the one I put in the wpa_supplicant when booting Buster 10 for the first time (didn't do this during these tests)
4. The PI was not accessible through the Wlan0 ip though so I did: 
5. `_sudo systemctl disable raspap.service_`
6. `_sudo reboot_`
7. AP was not in the air, however hostapd was up in the webgui (through OTG)
8. then I could _`change the setting to WiFi client AP mode (managed mode)`_
9. then I could see a full list of all AP's in my neighborhoud
10. then I restarted the service and I could connect to the raspi-webgui

then I did: 

1. `_sudo systemctl enable raspap.service_`
2. `_sudo reboot_`
3. And I could still see the raspi-webgui SSID in the air.
4. I connected to the raspi-webgui and my machine and the PI got internet (!)

So my premature conclusion of these findings is that it seems that the raspap service does something my PI doesn't like when managed mode is not enabled yet.

billz, 2019-12-22:
You need a managed mode AP during installation of... what exactly? Please clarify

dlucas9, 2019-12-23:
I have a similar issue and my observations are as follows:

If there is no client connection configured in the wpa_supplicant file, hostap runs fine but never advertises ...
You can force it to advertise by restarting the hostap service, but then any new clients will not connect until you reboot.

I have got things to work by creating a dummy client config in the wpa_supplicant file and all seems to be OK ... so far ... I think.

LeaTaka, 2019-12-23:
> You need a managed mode AP during installation of... what exactly? Please clarify

- Regarding this topic: I need managed mode to be enabled during installation so I don't have to do this manually after installation.
- Regarding my project: I'm testing a babymonitor on the PI zero and need to frequently reinstall all software. Maybe a bit overengineered but the more I can automate this process the less time I loose during development.

billz, 2019-12-23:
@LeaTaka you would probably be better off using a container or one of many code deployment tools. 

Your original bug report was that the managed mode AP button was not enabled without a configured wireless client. This is by design. You are welcome to fork this repo and modify it for your needs or use a previous release of RaspAP, but I'm unable to assist you with troubleshooting your project.

closed 2019-12-23 by billz

LeaTaka, 2019-12-23:
You are right, I need to learn more about deployment option.

Ref. the design: fair deal. I will use a workaround like fr example dlucas9 mentioned. 

Thanks for the support and this great piece of software!