RaspAP/raspap-webgui #460: config.php not found

flubbe, 2019-12-18:
**Describe the bug**
The graphs and numbers on the bandwidth page do not show.

**Your environment**
- Raspberry Pi 4 with Raspbian Buster Lite. 
- RaspAP quick install

**Expected behavior**
The graphs and numbers on the bandwidth page show 

**Additional Information**
The lighttpd log file shows that "config.php" cannot be found from [index.php](https://github.com/billz/raspap-webgui/blob/32d398f07d1f98a2a52d8e3905c0250aacafb12e/index.php) and [get_bandwidth.php](https://github.com/billz/raspap-webgui/blob/32d398f07d1f98a2a52d8e3905c0250aacafb12e/ajax/bandwidth/get_bandwidth.php) .

**Proposed solution**
The [config.php](https://github.com/billz/raspap-webgui/blob/master/config/config.php) file included in 

https://github.com/billz/raspap-webgui/blob/32d398f07d1f98a2a52d8e3905c0250aacafb12e/index.php#L24

and

https://github.com/billz/raspap-webgui/blob/32d398f07d1f98a2a52d8e3905c0250aacafb12e/ajax/bandwidth/get_bandwidth.php#L5

should possibly included as 'config/config.php' and '../../config/config.php', respectively.

billz, 2019-12-20:
During installation, config.php is copied from /config to /includes. See https://github.com/billz/raspap-webgui/pull/426

billz, 2019-12-30:
Closing due to inactivity. Reopen if needed, thanks

closed 2019-12-30 by billz