RaspAP/raspap-webgui #462: 8 users/connection allowed?

agoralive, 2019-12-23:
Is this the maximum allowed users?

Is there a way to expand this?

billz, 2019-12-23:
RaspAP gives you control of hostapd's `max_num_sta` setting. As the help text clearly states, 'The default and maximum is 2007'.

agoralive, 2019-12-25:
I have 8 units connected to the AP, and then it allows no more.

Connection is just refused for extra devices, and I have this 2007 in the max_num_sta, so definitely not 2007 allowed connections?

Or is there something I just don't understand?

Merry Christmas by the way :)

agoralive, 2019-12-25:
I have tried to set the client number to 2007, 100 and 10

No difference, it allowes only 8 connections :(

billz, 2019-12-25:
The bottleneck is likely not hostapd, but rather a limitation of the brcmfmac driver. Other RPi users have reported substantially greater maximum AP client connections, so I'd hesitate to say it's a hardcoded limitation of brcmfmac. Available RAM, under voltage and other factors come into play. You don't mention which RPi you're using, but I would not, for example, expect more than this from a Pi Zero W.

Related discussion here https://github.com/raspberrypi/linux/issues/3010

Merry Christmas :)

agoralive, 2019-12-25:
I am using a Raspberry Pi 3B+

billz, 2019-12-26:
Have you tried running hostapd in debug mode while you connect clients?

```
sudo hostapd -dd /etc/hostapd/hostapd.conf
```

agoralive, 2019-12-26:
Had not tried, I have now. This only throws everything off and disables wlan0.

No I will "just" live with these 8 clients, was just hoping for a way to connect 16-20 devices....

Maybe I will look into the esp_wifi_repeater, to see if this can do something....

agoralive, 2019-12-26:
I know I closed this, but would more connections be possible if I added an USB Wifi Adapter?

billz, 2019-12-30:
It would not hurt to test with an external adapter. If you're able to connect >8 clients with the same kernel and hostapd config, it's reasonable to infer that the bottleneck is at the Broadcom WiFi chip supported by the brcmfmac driver.

billz, 2020-01-06:
Closing this, feel free to reopen if you have more info to add.

closed 2020-01-06 by billz

agoralive, 2020-01-16:
Just as a comment, adding another wifi adapter to the Rpi, did nothing :(

Does anybody know how many the Rpi 4 allows to have connected?

ezimmert, 2020-02-19:
I have just conducted tests today with a Rpi4 4Gb RAM and I wasn't able to connect more than 8 devices. Same with the 3b+. Both were running latest version of Buster.

I conducted the same tests 1 year ago and the results were quite different with Stretch:
19 users with a Rpi 3b and 14with a Rpi 3b+...

billz, 2020-02-20:
@ezimmert interesting data point, thanks for reporting. This may indeed be the upper bound set by brcmfmac. Out of curiosity, what is your kernel + brcmfmac firmware version?

In an ideal world, broadcom/cypress would release the firmware source code so we could verify this directly.

andrew-d-l-nelson, 2020-10-21:
Is this something that has been addressed? I am using a Rpi4 4GB RAM with Buster and have been unable to get past 7 clients on my hotspot. It isn't a resource issue on the Rpi itself it seems (ram and cpu doesn't seem to be affected). Would love to have more able to connect at once. My workaround has been to have Rpi clients turn off their WiFi for 5 minutes ever hour to let other Rpis get a spot on the router to transfer their data. Less than optimal :D

guillain, 2020-12-19:
Hi,

Same issue but I've added my 2 cents in the [issue #3010 ](https://github.com/raspberrypi/linux/issues/3010#issuecomment-748450967) (still open ^^)
=> max 8 clients

Have fun and take care