RaspAP/raspap-webgui #464: Cannot access Pi by hostname

EricPoggel, 2019-12-25:
* [x] This is a bug report
* [X] I searched existing issues before opening this one
* [X] I checked the FAQ before creating this issue
* [X] I have read and understand the issue reporting guidelines 

**Describe the bug**
Cannot access Pi by hostname from client connected to access point.

**Your environment**
Pi 4 1GB with Raspbian Buster Desktop.  
Quick install with all default options except install folder of /home/pi/raspap/
Onboard wireless.
internet provided via ethernet cable.

**Steps to reproduce**
Connect a windows 10 computer to the access point.
ping 10.3.141.1 // works
ping raspberrypi // fails
ping raspberrypi.lan // fails

Connect a windows 10 computer to the same network as the ethernet cable, and all of the pings above work.

billz, 2019-12-25:
This project does not modify the default hostname, nor the avahi-daemon which provides mDNS.

closed 2019-12-25 by billz

EricPoggel, 2019-12-26:
Well I figured out how to do it, just in case anyone else wants to do the same.  I modified dnsmasq.conf like so:

`interface=wlan0
dhcp-range=10.3.141.50,10.3.141.255,255.255.255.0,12h

# Tell our clients to use us as a DNS server:
dhcp-option=6,10.3.141.1

# Prevent using the OS's DNS server
no-resolv

#Set the upstream DNS server to use:
server=1.1.1.1

# Allow typing http://raspberrypi in the browser to access this server.
address=/raspberrypi /10.3.141.1`