RaspAP/raspap-webgui #467: Web server option at install (nginx or apache instead of lighttpd)?

Thatoo, 2020-01-01:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [ ] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Would it be possible to install RaspAP with nginx web server instead of lighttpd?

**Your environment**
* Raspberry Pi 4
* Raspbian version ( Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install 
* Using default configuration? Y
* Simultaneous AP and managed mode? N
* Onboard wireless chipset 
* Other software or services running with RaspAP? No but would like to have nginx

**Expected behavior**
Be able to install RaspAP with nginx instead of lighttp as I would like to install other apps (most use nginx nowadays) such as Lufi on my Raspi4 (https://framagit.org/fiat-tux/hat-softwares/lufi)

**Actual behavior**
if I install RaspAp with the fast install and then nginx, it creates confilct which seems logical as RaspAP installs lighttpd

billz, 2020-01-01:
Have you tried changing lighttpd's default port on the System tab? This feature was added to enable compatibility with other web servers.

Thatoo, 2020-01-01:
as explain here https://github.com/billz/raspap-webgui/wiki/FAQs#webport ? no, I've searched for nginx and apache key words and I found this issue https://github.com/billz/raspap-webgui/issues/294 but not this FAQ. I'll have a look at this option then and I'll come back to you.

Thatoo, 2020-01-02:
I can indeed install nginx and apps that I can reach with port 443 (https) and my own namedomain with Let's encrypt certificate but then I can't reach raspap this way anymore but only with http on port 8080.
Good by https://raspa.mydonain.ltd option with Let'sencrypt certificate (thank you certbot) as I did...

Being able to have raspap on nginx would be quiet nice (to have all webapp accessible on 443 and https) but I understand that it's not a top priority...

billz, 2020-01-02:
The Quick Installer prompt 'Enter alternate Lighttpd directory' can be used to install this project into an arbitrary location, including an nginx or apache web root.

billz, 2020-01-02:
Closing this as installing alternate web servers is beyond the scope of this project.

closed 2020-01-02 by billz