RaspAP/raspap-webgui #468: No means to automatically read and use  VPN servers from vpngate.net.

SecuritasMachina, 2020-01-01:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x ] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Lack of functionality to use as travel router.  User has to establish connection and then download & upload a VPN configuration file before using OpenVPN functionality.  This will be a pull request to provide a baseline for said behavior.

**Your environment**
* Raspberry Pi hardware = Model 4B v1.1
* Raspbian version (examples: Buster Lite, Buster Desktop) = Buster Desktop
* Followed the project prerequisites? (Y/N) = Not sure.
* Checked the project FAQ? (Y/N) = Yes
* RaspAP Quick Install or Manual setup? = Quick Install, have to manually setup to keep going, I'll figure out how to fix and contribute later.
* Using default configuration? (Y/N) = No
* Simultaneous AP and managed mode? (Y/N) = Not yet
* Onboard wireless chipset or external adapter? = Both
* Other software or services running with RaspAP? = Enhanced DNSMasq, otherwise stock

**Steps to reproduce**
Tell us how to reproduce this issue. Provide as much detailed information as possible.

**Expected behavior**
Goto OpenVPN Section.  Click get VPNs, click VPN you wish to use, click start VPN.

**Actual behavior**
N/A  behavior doesn't currently exist.

**Screenshots**
N/A

**Additional context**
I'm lazy and don't want to bounce around unsecured on my devices.  I'm using a micro display on my RaspBerryPI.

billz, 2020-01-02:
This is a feature request for... populating a list of VPN servers from a third-party website?

closed 2020-01-04 by billz

SecuritasMachina, 2020-01-04:
Yes, I already coded it & updated install w/ needed files.