RaspAP/raspap-webgui #470: Configure hotspot: Channel always shows 1

PhilipPeake, 2020-01-03:
**Describe the bug**
On the "Configure hotspot" screen, the Channel always shows as "1". Changing the value and saving the settings does change the channel in the config file (takes effect on next reboot). Even after reboot, and on new channel, still displays "1" as the channel.

**Your environment**
* Raspberry Pi 4
* Raspbian Buster desktop
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Y
* Using default configuration? (Y - except no VPN)
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter?  Onboard
* Other software or services running with RaspAP? None

billz, 2020-01-04:
What is the output of /ajax/networking/get_channel.php ?

billz, 2020-01-04:
Thanks for reporting

commit 6106c319 (2020-01-04): Bugfix. Resolves #470
closed 2020-01-04 by billz