RaspAP/raspap-webgui #471: Simultaneous AP and wifi client mode: AP button cannot be enabled

feh123, 2020-01-04:
* [x] This is related to #452 but new because previously I have been able to activate wifi client AP mode
* [x ] I searched existing issues before opening this one
* [ x] I checked the FAQ before creating this issue
* [x ] I have read and understand the issue reporting guidelines 

**Describe the bug**
Try to activate managed mode but cannot set AP button to enabled. 

**Your environment**
* Raspberry Pi hardware: Pi 3 Model B+  
* Raspbian version: Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset 
* Other software or services running with RaspAP? No

**Steps to reproduce**
Add wpa_supplicant file for my nerwork wifi and ssh to my sd card (after flashing Buster Lite with Etcher) and operated the RPi headless with no eth0. Installed as above. Edited config.php and dhcpcd.conf. 
Opened RaspAP UI in browser. Wlan0 and hostapd were both down. On wifi client AP mode page I tried to enable AP mode. Button stayed on disabled.
Repeated the above three times with the same result.  
Any thoughts about a possible cause appreciated.
**Expected behavior**
I expected to enable the button
**Actual behavior**
Button remains disabled.
**Additional context**
I did get RaspAP v2.0 button to enable.

billz, 2020-01-04:
Unable to repro this on a headless 3B+ connected to a local wireless network with wpa_supplicant.

> Try to activate managed mode but cannot set AP button

this feature does not activate managed mode; it enables an AP _after_ you have configured the Pi as a wireless client. Refer to the project readme. You can either do this manually with wpa_supplicant, as described in the FAQ, or with the Configure Wifi client UI. 

> Wlan0 and hostapd were both down

if wlan0 is down, you do not have a configured wifi client interface. The link quality graph on the dashboard is an indicator. see above.

PhilipPeake, 2020-01-04:
I see the same thing on my R-PI 4B.
I have configured  wpa_supplicant.conf identically to the way it was (working) before installing.
The Client is not working. When I go to activate it, the slider is inactive (greyed out).

Maybe related: If I try to configure using raspi-config, it errors out with  "Could not communicate with wpa_supplicant".

wpa_supplicant is running:

# ps -fe | grep wpa
root       333     1  0 14:12 ?        00:00:00 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant

billz, 2020-01-05:
> If I try to configure using raspi-config, it errors out with "Could not communicate with wpa_supplicant". wpa_supplicant is running

It may be running, but has a socket error. Refer to [this FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#scanning). This appears to be an issue since Buster's release. Some users have solved it by [downgrading wpa_supplicant](https://www.raspberrypi.org/forums/viewtopic.php?t=244731).

feh123, 2020-01-05:
Hi

Thanks for the fast response. I have it working now. As you can guess I have some problems when I try AP mode. In AP mode with only one wifi I just need to be very careful to follow your instructions - so my mis-reading is the issue.
But AP mode and one wifi does not work well for me. Not all my devices attach. So not a RaspAP problem but I would be interested if others have seen this. If I use #448 I can get a bridge to work and all my devices then attach. But #448 means you are not really using RaspAP as a UI anymore.
Thanks again.

closed 2020-01-05 by feh123

billz, 2020-01-05:
If you want to revert to the v2.0 behavior, edit the following:

https://github.com/billz/raspap-webgui/blob/6106c3196aa5d8aea735d277dd7b110ae5a04bea/templates/hostapd.php#L133

removing `<?php echo $wifiAPToggle; ?>`

feh123, 2020-01-06:
Hi

Thanks for this. It did not seem to effect my AP. However in trying it out I noticed that my AP only broadcasts at 5Ghz. The client wifi signal 5Ghz and my hotspot seems to keep this value. I am guessing my device only handles 2.4Ghz. I tried rescanning to find the 2.5Ghz signal from my network wifi but rescanning did not find it. Is there a way to force 2.4Ghz?