RaspAP/raspap-webgui #473: Not able to connect to internet via hotspot

colindawson, 2020-01-06:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
devices connected to the hotspot are unable to browse the internet.

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+, Pi Zero W)    PI 4 ($Gb)
* Raspbian version (examples: Buster Lite, Buster Desktop) Buster installed 6-Jan-2020
* Followed the project prerequisites? (Y/N) Y
* Checked the project FAQ? (Y/N) Y
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? (Y/N)     No to OpenVPN all others Yes.
* Simultaneous AP and managed mode? (Y/N) Yes
* Onboard wireless chipset or external adapter?  On board only
* Other software or services running with RaspAP?  None

**Steps to reproduce**
I followed this instructional video https://www.youtube.com/watch?v=YbvSS8MJm2s&t=828s
every step except for the one about doing a change to a file at timestamp 8:00 (as this has been fixed at install time)

Connected to working WIFI network  (tried both Hotspot on Phone and my home network)

Hotspot settings

Basic -
Interface: uap0
SSID: raspi-webgui
Wireless Mode: 802.11g - 2.4 GHz
Channel: 1

Security -
Security Type: WPA2
Encryption Type: CCMP
PSK: ChangeMe

Advanced:
Enabled - WiFi client AP mode
disabled - logfile output
disabled - Hide SSID in broadcast

Maximum number of clients: 2007
Country Code: United Kingdom

**Expected behavior**
I should be able to browse to http://raspberrypi.org when using an device connected to the hotspot.

**Actual behavior**
The device can connect to the PI and VNC into 192.168.50.1
Chromium running on the PI can connect to the internet
Browser on the device (iPad in my case) cannot browse to any dns address.
Browser on device can access an IPAddress directly through the hotspot.

It appears that DNS Lookups are working correctlt

**Screenshots**

**Additional context**

billz, 2020-01-07:
Cannot reproduce with the information provided. I did not view the video you referenced.

closed 2020-01-07 by billz

colindawson, 2020-01-08:
Turns out that the root cause of the issue is that the default install of Raspbian is configured not to wait for network on boot.   Turning this setting on resolves the issue.   If the setting is off, once booted you'd need to stop and start DNSMASQ to allow the dns lookups to start working.

billz, 2020-01-08:
Glad to hear you got it sorted. Out of curiosity, did you have the raspap.service enabled?

colindawson, 2020-01-09:
Hi bill

I think the device was enabled.  I didn’t check, but had not done anything to change it.

Regards

Colin