RaspAP/raspap-webgui #474: Managed Mode AP - Scanning not working and configure hotspot limited

feh123, 2020-01-07:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
I cannot change wifi client, cannot alter hotspot configuration.

**Your environment**
* Raspberry Pi hardware: Pi 3 Model B+
* Raspbian version: Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install Y
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset Y 
* Other software or services running with RaspAP? N

**Steps to reproduce**
Follow quick installer and Manged Mode procedure with wpa_supplicant file and ssh file added to sd card at boot. I checked I have the lastest master. I have repeated this procedure twice with different sd cards.

**Expected behavior**
I want to pick a 2.4Ghz wifi client that is available from my router. 

**Actual behavior**
Wifi client scan only finds one wifi - the 5Ghz wifi and connects to this band. It does not find the router's 2.4Ghz band. The scan finds other (neighbours wifi's) but all these are ghosted. If I pick a 5Ghz mode from the list on the hotspot configure page I cannot alter the channel to 36 - it stills shows a limit of 12
If I use a wifi app to check my RPi hotspot it only shows the hotspot on the 5Ghz channel 36. I guess this is forced by the wifi client?
If I check what bands are present using sudo wpa_cli -i wlan0 scan and sudo wpa_cli -i wlan0 scan_results I do see the 2.4Ghz band.

**Screenshots**
**Additional context**
The 5Ghz band works fine and I can connect my phone, tablet etc and get an internet connection. But some devices only use the 2.4Ghz band.

closed 2020-01-07 by billz

feh123, 2020-01-07:
Hi

It is possible that I am not following the procedure re managed mode correctly for sure. I am just not sure what part of the FAQ I am mis-understanding. I have tried many ways to configure the hotspot including carefully stopping and restarting the hotspot. If there is something I am missing I cannot see it (and it could well be the obvious). I have a "normal" RaspAP using eth0 and that gives me a 2.4Ghz wifi that works perfectly. I also have a wlan0/wlan1 RaspAP that works fine and gives me a 2.4GHz wifi. I like the software a lot.
The irony is that I quite like the idea of a 5Ghz wifi - it's just this time I need a 2.4 band! Any thoughts much appreciated!

billz, 2020-01-13:
The RPi 3b+/4's Cypress wifi chipset is capable of operating on either the 2.4 or 5 GHz band, but does not support real simultaneous dual-band (RSDB). This means if you want to switch to hosting a managed mode AP on the 2.4 GHz band, you must (re)configure the wireless client to use the same band.

Wireless routers that support RSDB usually have at least two onboard wifi chips.

feh123, 2020-01-13:
Hi 
Thanks for the info. I have tried to re-configure wireless client via the wpa_supplicant file to use the 2.4Ghz band but without success. So I have gone back to my 3B RPi and then every is fine. Thanks again.