RaspAP/raspap-webgui #476: Changing hotspot ssid and password causes raspap to not be able to connect anymore

jtcrank, 2020-01-11:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
After having a working installation, I change the ssid and password of the hotspot in the webgui. Upon reboot the ssid is broadcast, but devices will not connect despite having the proper password. (I checked the hostapd config to be sure). I have started fresh and followed the same procedure with the same results. To be sure rebooting wasn't an issue, before changing the ssid and password again, i did a couple reboots to make sure I could connect, was successful each time. Only when changing ssid and password does something go wrong.

**Your environment**
* Raspberry Pi 4 (4gb)
* Raspbian Buster Lite
* Followed the project prerequisites? (Y
* Checked the project FAQ? (Y)
* RaspAP Quick Install
* Using the FAQ solution to wan0 AP and wan1 client
* Onboard wireless chipset for wan0 AP and RT5370 USB dongle 
* Raspap set up on fresh installation. No additional software has been used.

**Steps to reproduce**
After having a working connection to the raspap web gui from another device, go to the hotspot tab and change both the ssid and the password, save and reboot. I changed no other settings.

**Expected behavior**
The new SSID shows up and I should be able to connect with a proper password.

**Actual behavior**
The new ssid shows up but a proper password does not connect.

**Additional context**
- Manually changing the ssid and password back to the default in the hostapd.conf file does not re-establish a working raspap.
- checking hostapd in system logs shows that device is associated, passes handshake, then immediate disassociated.

billz, 2020-01-11:
> Be aware that external WiFi adapters (ie, USB "dongles") vary greatly in terms of hardware capabilities and driver support. Many do not work out of the box on the RPi, require a powered USB hub, manual driver and/or firmware installation or are otherwise not well suited for the RPi. For these reasons, issues related to wlan0 and wlan1 configurations with external adapters will not be considered.

closed 2020-01-11 by billz

jtcrank, 2020-01-12:
@billz I did not think this was an issue with the usb dongle so I apologize. I did however find out what is causing the issue on reboot. Whenever the system reboots, the `/etc/dhcpcd.conf` file reverts back to what it was before making the changes described in the FAQ.
I update it to:
```
interface wlan0
static ip_address=10.3.141.1/24
static domain_name_server=1.1.1.1
nohook wpa_supplicant
```
but each time I edit the hotspot configs, save, and reboot it goes back to:
```
interface wlan1
static ip_address=10.3.141.1/24
static routers=10.3.141.1
static domain_name_server=1.1.1.1 8.8.8.8
```

Would you still consider this behavior related to the usb dongle itself or the software in regards to using another interface in general alongside wlan0?

feh123, 2020-01-12:
Hi

I have tried to use a wlan0, wlan1 setup - I have only succeeded by significantly modifying RaspAP using a procedure from @up2smoke. It creates a bridge. For wlan1 I used a Linkysis WUSB6300. There are a few threads discussing the dual wifi idea because as @billz says the dongle is a problem as the chipset needs to have several key elements and chipset documentation is often very poor. There are lists of raspberry pi working dongles but I am not sure they are always reliable. So I can see it's something outside of RaspAP's contriol. 
I am interested in trying to help fix the the many issues (I think the threads cover many but may not be complete!) because RaspAP is very useful software for me and the flexibility of two wifi's is very attractive.

billz, 2020-01-12:
@jtcrank per the FAQ, this config is only _partly_ supported by RaspAP. Your setup is reverting to this project's [default configuration](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#default-settings). There is no explicit support for external adapters, bridging interfaces or wlan0/wlan1 configs, nor are there plans for this. 

As a developer, I had to draw the line at supporting the RPi's onboard wireless only. It is not practical (or even possible) for me to support features that lead to troubleshooting everyone's unique configs and add-on hardware. I hope this makes sense.

Animus3, 2020-04-09:
I had the same problem. I'm using a raspberry pi 4 and an external USB high gain adapter. External adapter as a client and internal as hostpad. When I change SSID or Password, I can't connect anymore, my devices still looking for an IP. I uninstalled and reinstalled raspAP; then I change SSID and Password editing /etc/hostapd/hostapd.conf as root, and It works.

feh123, 2020-04-10:
Hi can you say more about the steps you used to get raspap working? This is my usual problem - with a Linksys USB wifi as wlan1 - devices are not given an IP address. Thanks