RaspAP/raspap-webgui #477: Multiple ssid with one interface

samWieczorek, 2020-01-16:
Hi
Is it possible with raspAP to configure two AP (two ssid) on the same interface. This feature is available in hostapd.

Thanks

billz, 2020-01-17:
According to the output of `iw list`:

```
valid interface combinations:
    * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
    total <= 3, #channels <= 2
    * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
    total <= 4, #channels <= 1
```
the RPi's onboard wireless chipset is only capable of hosting a single AP. You can host a second AP on an external WiFi adapter, however this is beyond the scope of this project.

billz, 2020-01-24:
Closing due to inactivity. Reopen if needed, thanks

closed 2020-01-24 by billz