RaspAP/raspap-webgui #478: Proposed enhancement - CPU Temperature

Weylin-fr, 2020-01-20:
I miss the Temperature of my Raspberry, till it's closed in a case without a fan

I did'nt add the translations
see my pull request for more information

[image]

billz, 2020-01-20:
Nicely done :) I will check out your PR, thanks!

billz, 2020-01-22:
I can see a case for having mem, cpu load & temp stats in a small widget on the dashboard. Easily done with the System class.

Weylin-fr, 2020-01-22:
ok, why not
Do you want to do that ?

billz, 2020-01-22:
Maybe, but after we incorporate your changes. I still think it belongs here on the System page. I made a small fix and opened a PR on your branch.

billz, 2020-01-22:
@Weylin-fr merged with master. nice addition, thank you :)

closed 2020-01-24 by billz