RaspAP/raspap-webgui #480: Iptables help flask

kwaknet01, 2020-01-22:
I have a simple CRUD flask app which runs localhost:5000 for wifi users, my RaspAP is configured with nodogsplash. I'm having problem with how can i access my app within the hostpot itself but i can access it outside as long as i'm in a same network. i have zero knowledge about iptables. hehe
i've also read issue #333 i've tried it but still i can't access it. maybe i was doing it wrong. 

It would be a great feature if RaspAP has CRUD template on it :)

my configuration /etc/rc.local

`echo 1 > /proc/sys/net/ipv4/ip_forward`
`sudo iptables -t nat -A POSTROUTING -j MASQUERADE`
`sudo iptables -t nat -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE`

#sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
#sudo iptables -A FORWARD -i eth0 -o wlan0 -m state --state RELATED,ESTABLISHED$
#sudo iptables -A FORWARD -i wlan0 -o eth0 -j ACCEPT
#sudo iptables -A INPUT -j ACCEPT
#sudo iptables -A OUTPUT -j ACCEPT
#sudo iptables -I INPUT -p tcp --dport 5000 -j ACCEPT

`sudo nodogsplash &`
`exit 0`

billz, 2020-01-30:
You need a rule to forward traffic from the AP side (wlan0) to your app. One way to achieve this is with NAT prerouting. Assuming the flask host is a public server:

`iptables -t nat -A PREROUTING -p tcp --dport 80 -i wlan0 -j DNAT --to 0.0.0.0:5000`

You can get more insights into what traffic is doing while connected to the AP with `tcpdump -n -i wlan0 port 5000`.

This is just a starting point. As I don't have an environment like what you describe, I can't really advise you beyond this. Best of luck.

closed 2020-01-30 by billz

kwaknet01, 2020-06-25:
hello i know this is kinda old thread. I've solved this problem by adding a rule at /etc/nodogsplash/nodogsplash.conf.

`FirewallRuleSet users-to-router {`
 `  FirewallRule allow tcp port 5000`
`}`

seantanjx, 2021-12-18:
Hi @kwaknet01, how do you run your own flask app as a captive portal on NDS? I placed my flask app in the /etc/nodogsplash/htdocs folder but I have no idea how to start the flask server. Could you advise? Thanks!