RaspAP/raspap-webgui #481: OpenVPN not Working

Realtox, 2020-01-23:
* [x] This is a bug report
* [ ] I searched existing issues before opening this one
* [ ] I checked the FAQ before creating this issue
* [ ] I have read and understand the issue reporting guidelines 

**Describe the bug**
Wen i upload the .ovpn file, he say "uploaded the OpenVPN sucessfully" but it dont make a OpenVPN Hotspot.

**Your environment**
* Raspberry Pi hardware (Pi 3 Model B+
* Raspbian version (Raspbian)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Manual
* Using default configuration? (Y)
* Simultaneous AP and managed mode? Managed
* Onboard wireless chipset or external adapter?  Onboard
* Other software or services running with RaspAP? No

**Steps to reproduce**
I Uploadet the .ovpn File from my OpenVPN Server.

**Expected behavior**
I would make a Hotspot, that runs with this OpenVPN Profil, that alle Devices is in this VPN and dont Need the OpenVPN Program.

**Actual behavior**
Now i have a normaly Hotspot without a VPN Service.

**Screenshots**
I dont have one.

billz, 2020-01-23:
Per the FAQ:
> If OpenVPN fails to start, check for errors with `sudo systemctl status openvpn-client@client` and `journalctl --identifier openvpn`

Realtox, 2020-01-24:
I have check that, but it say, it running.

billz, 2020-01-24:
Cannot reproduce with the info provided.

closed 2020-01-24 by billz

zhampu, 2021-06-11:
`sudo systemctl status openvpn-client@client
● openvpn-client@client.service - OpenVPN tunnel for client
   Loaded: loaded (/lib/systemd/system/openvpn-client@.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-06-11 20:52:07 CEST; 8min ago
     Docs: man:openvpn(8)
           https://community.openvpn.net/openvpn/wiki/Openvpn24ManPage
           https://community.openvpn.net/openvpn/wiki/HOWTO
 Main PID: 516 (code=exited, status=1/FAILURE)

Jun 11 20:52:07 raspberrypi openvpn[516]: Options error: --cert fails with 'client.crt': No such file or directory (errno=2)
Jun 11 20:52:07 raspberrypi openvpn[516]: Fri Jun 11 20:52:07 2021 WARNING: cannot stat file 'client.key': No such file or directory (errno=2)
Jun 11 20:52:07 raspberrypi openvpn[516]: Options error: --key fails with 'client.key': No such file or directory (errno=2)
Jun 11 20:52:07 raspberrypi openvpn[516]: Fri Jun 11 20:52:07 2021 WARNING: cannot stat file 'ta.key': No such file or directory (errno=2)
Jun 11 20:52:07 raspberrypi openvpn[516]: Options error: --tls-auth fails with 'ta.key': No such file or directory (errno=2)
Jun 11 20:52:07 raspberrypi openvpn[516]: Options error: Please correct these errors.
Jun 11 20:52:07 raspberrypi openvpn[516]: Use --help for more information.
Jun 11 20:52:07 raspberrypi systemd[1]: openvpn-client@client.service: Main process exited, code=exited, status=1/FAILURE
Jun 11 20:52:07 raspberrypi systemd[1]: openvpn-client@client.service: Failed with result 'exit-code'.
Jun 11 20:52:07 raspberrypi systemd[1]: Failed to start OpenVPN tunnel for client.`