RaspAP/raspap-webgui #484: Requesting Management Frame Protection (MFP) support

antgiant, 2020-01-26:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [ ] This is a bug report
* [x] I searched existing issues before opening this one
* [x] I checked the FAQ before creating this issue
* [x] I have read and understand the issue reporting guidelines 

**Describe the bug**
Requesting Hostapd configuration support for 802.11w. (aka management frame protection (MFP).) This would allow for the RaspAP wireless networks to be protected against deauth attacks.

**Your environment**
* Raspberry Pi 4 
* Raspbian version Buster Desktop
* Followed the project prerequisites? Yes
* Checked the project FAQ? Yes
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Y
* Onboard wireless chipset or external adapter? Both
* Other software or services running with RaspAP? No

**Steps to reproduce**
It is not currently possible to enable management frame protection from within the RaspAP configuration screens. 

**Expected behavior**
hostapd management screens allow for turning it on or default it to always on.

**Actual behavior**
hostapd defaults to it off, and it is not a configurable option.

**Screenshots**
N/A

**Additional context**
Since this prevents deauth attacks and generally improves the security of the network I would like to request that it be configurable, or simply turned on at all times. Managing only requires looking at two additional lines in the hostapd.conf file. Here is the relevant snip from the complete 
[hostapd.conf options](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf)
`# ieee80211w: Whether management frame protection (MFP) is enabled
# 0 = disabled (default)
# 1 = optional
# 2 = required
#ieee80211w=0
# The most common configuration options for this based on the PMF (protected
# management frames) certification program are:
# PMF enabled: ieee80211w=1 and wpa_key_mgmt=WPA-EAP WPA-EAP-SHA256
# PMF required: ieee80211w=2 and wpa_key_mgmt=WPA-EAP-SHA256
# (and similarly for WPA-PSK and WPA-PSK-SHA256 if WPA2-Personal is used)
# WPA3-Personal-only mode: ieee80211w=2 and wpa_key_mgmt=SAE`

billz, 2020-01-27:
I've done some testing with 802.11w on the RPi 4.

**The good news:** The Cypress WiFi chipsets used by the RPi 3b+/4 ([CYW43455](https://www.cypress.com/file/358916/download)) and Pi Zero W ([CYW43438](https://www.cypress.com/file/298076/download)) both support IEEE 802.11w. The brcmfmac kernel driver supports MFP when the device/chipset supports it. Indeed, starting hostapd (v2.8 Buster) with these settings:

```
ieee80211w=2
wpa_key_mgmt=WPA-EAP-SHA256
```

successfully starts an AP with MFP enabled. Network security is broadcast as WPA3 Enterprise. 

**The not-so-good news:** iOS 13 fails to connect to WPA3 networks. Despite [claims to support it](https://www.apple.com/ios/ios-13/features/), this appears to be a bug with Apple's implementation. I've tried the latest release (13.3) on multiple iOS devices with no joy. There's an [active thread](https://forum.openwrt.org/t/hostapd-2-7-2-9-wpa2-wpa3-mixed-ios13-failure/45225) on the OpenWRT forums about this. 

The WiFi alliance left the decision about implementation to the hardware vendors. A vendor can add support via a software update or leave it for new hardware. I have not (yet) tested on Android or other devices.

In practice, I think it will take some time before we see broad vendor support for WPA3.

antgiant, 2020-01-29:
Thanks so much for the super fast response. I now have also tried a few iOS 13 devices and cannot get them to work. :-(
Here is hoping Apple fixes this soon.

billz, 2020-03-13:
In the meantime, there may be a case for adding an "experimental" 802.11w option to the hostapd config.

hvenev, 2020-06-06:
@billz How did you get MFP to work on CWY43455? For me hostapd fails to start with the following error appearing in the kernel log:
```
ieee80211 phy1: brcmf_cfg80211_add_key: set wsec error (-52)
```

billz, 2020-06-06:
@hvenev it worked with whichever kernel I had installed in January, fails now with the same error.

Simon-Davies, 2020-11-17:
Any update on this?

billz, 2022-01-04:
Dredging up an old request here. 802.11w appears to have support from the 5.10.x kernel. Same hardware as before (RPi 4b / Cypress CYW43455).

802.11w enabled:
```
ieee80211w=1
wpa_key_mgmt=WPA-PSK

RSN Capabilities: 0x008c
    .... .... .0.. .... Management Frame Protection Required: No
    .... .... 1... .... Management Frame Protection Capable: Yes
```

802.11w required:
```
ieee80211w=2
wpa_key_mgmt=WPA-PSK-SHA256 

RSN Capabilities: 0x008c
    .... .... .1.. .... Management Frame Protection Required: Yes
    .... .... 1... .... Management Frame Protection Capable: Yes
```

Kernel version `Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux`

closed 2023-03-30 by billz