RaspAP/raspap-webgui #485: Limited privileged user in WEB-GUI

olewsaa, 2020-01-28:
**Request for enhancement**
Add a user in addition to admin, with access only to scan and select remote base station for wlan1 client when using wlan1 as a client and wlan0 & eth0 as servers. I am using the access point in a boat where crew want to be able to login on and scan and set the remote shore station. Giving full access with the admin account might break current manual setup. Hence limit the access to only selecting hot-spot ashore and entering password the it is a desired feature. 

**Your environment**
* Raspberry Pi hardware Pi 3 Model B+  
* Raspbian version is Buster Desktop
* Followed the project prerequisites? Yes
* Checked the project FAQ? Yes
* RaspAP Quick Install ? Quick install
* Using default configuration? No, some manual steps to make wlan1 (long-range USB antenna) as client and wlan0 and eth0 as server. 
* Simultaneous AP and managed mode? No.
* Onboard wireless chipset or external adapter? Both, wlan1 as client, wlan0 and eth0 as server. 
* Other software or services running with RaspAP? None

**Steps to reproduce**
No steps are needed to include a limited privileged user.  My setup is found at: 
https://sites.google.com/site/olewsaa/yacht-server/raspberry-pi-as-a-router-gateway  and 
https://github.com/olewsaa/Yacht-computer/tree/master/wifi2wifi/web-version 
It works very well, with the exception that any crew member logged in a admin can overwrite the configuration and trigger a tedious manual reconfiguration.  

**Expected behavior**
The user login should only be allowed to do "scan" and select client connection and enter password for remote hot-spot. 

**Actual behavior**
Currently admin user can change anything and hence overwrite the config files added manually to set the access point to the desired setting.

billz, 2020-01-28:
Very interesting use case! From your description, enabling [monitor mode](https://github.com/billz/raspap-webgui/wiki/FAQs#monitor) could be an alternate solution. This allows all configuration actions of a wireless client (scan, update, connect, delete) but restricts the ability to administer any other services.

olewsaa, 2020-01-29:
I am happy that you liked the use case, the commercial [redbox](https://www.mailasail.com/Communication/Red-Box) does this a few other things, but is quite costly. I wanted something similar with a nice web interface. It seems that I have found a nice alternative. 

I have done some testing and it seems to limit the web-gui by removing the save buttons. This is just fine for my usage. Hopefully the crew will not manage to break the settings now. I still think that two users admin and user should be implemented in the web-gui, but this is just a minor issue now.

billz, 2020-01-30:
I agree that there's a valid case for a user role with limited privileges. Will work this into a future update, thanks.

closed 2024-10-12 by billz