RaspAP/raspap-webgui #489: raspap not working after host reboot

peterdowg, 2020-01-30:
**Before submitting an issue**
Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues

* [x] This is a bug report
* [X] I searched existing issues before opening this one
* [X] I checked the FAQ before creating this issue
* [X] I have read and understand the issue reporting guidelines 

**Describe the bug**
This is a fresh clean install of the RaspAp, all defaults in place, nothing changed. The application works, but after rebooting the machine the web gui is unreachable, the SSID is not visible, there is nowhere on in the instructions on how to start the process manually using the terminal.

**Your environment**
* Raspberry Pi hardware (examples: Pi 3 Model B+)  
* Raspbian version (Buster Desktop, fresh install)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (whatever is default)
* Onboard wireless chipset.
* Other software or services running with RaspAP? None

**Steps to reproduce**
Run the Quick install, verify that it's running by loging into the web gui. Reboot the hardware raspberry pi, see that nothing works.

**Expected behavior**
After the machine reboot, the RaspAp should be up and running.

**Actual behavior**
The RaspAp is not running after the machine reboot.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

billz, 2020-01-30:
Unable to repro with a fresh Buster Desktop install on the Pi 3B+. The SSID appears with the default configuration and is able to be joined after several reboots. Please be sure to follow the project prerequisites. You can also enable logging for hostapd on the Advanced tab of Configure hotspot.

closed 2020-01-30 by billz

peterdowg, 2020-01-30:
Hi Bill. You closed my ticket and I still have the same issue. What prerequisites do you have in mind? I followed the instructions to the last dot. I see online that there are other people having the same issue, so I am not alone here. What is the command to restart all the processes when they die? I can't be re-running the installation script each time, because since yesterday this is what I had to do. This is a brand new installation, the PI is freshly created from the newest Buster disk image downloaded from Raspberry site with updates received as suggested in your steps, and your code is freshly installed from your installation script. The thing is NOT working. Don't just close the ticket, ask me for more information and I will provide it to you. Tell me what you need and where I can find it and will send it to you.

billz, 2020-01-30:
Prerequisites are here https://github.com/billz/raspap-webgui#prerequisites
Feel free to drop me an email if you like. Address is in index.php

peterdowg, 2020-01-30:
I found the problem. This happens when I switch to 5GHz network. It only allows me to do "802.11a" however, the option "802.11ac" is for whatever reason displayed but not selectable for me. What do I do to fix this? My hardware supports 5GHz since it is PI 3 B+ model. 2.4 802.11n is too slow. Thanks again.

billz, 2020-01-30:
Glad to hear it. 802.11ac is discussed in detail here https://github.com/billz/raspap-webgui/wiki/FAQs#80211ac

peterdowg, 2020-01-30:
Thanks Bill. That worked.